Genesis-Embodied-AI/genesis-world #377: How do I run on a Raspberry pi5 ?

4EverBuilder, 2024-12-28:
Can anyone run on genesis-world in a Rasberry pi5? (8GB memory)

Problem is NOT the memory or size of the machine. Rather it just does not complie because seems everything is geared for x86 as opposed to aarch64 ?

Even docker fails!

zhouxian, 2024-12-29:
We only tested on x64 and apple silicon.
Could you check first if taichi supports raspberry? I.e does pip install taichi work? If it does it should be good

4EverBuilder, 2025-01-01:
> We only tested on x64 and apple silicon. Could you check first if taichi supports raspberry? I.e does pip install taichi work? If it does it should be good

BTW, Apple Silicon on (Darwin Kernel Version 24.2.0) runs but doesn't show any gui

4EverBuilder, 2025-01-03:
Can you alternatively run it on AWS EC2?

zhenjia-xu, 2025-01-10:
Here is an example of showing GUI on Mac: [link](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py)

As to AWS EC2, you can use the docker: [link](https://github.com/Genesis-Embodied-AI/Genesis/tree/main?tab=readme-ov-file#docker)

closed 2025-03-06 by YilingQiao