Genesis-Embodied-AI/genesis-world #380: c++: fatal error: Killed signal terminated program cc1plus

zhangiser, 2024-12-29:
I'm build LuisaRender by 
```bash
cd genesis/ext/LuisaRender
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF 
cmake --build build -j $(nproc)
```
The first cmake command run success
```bash
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF 
```
But the second cmake command run failed
```bash
cmake --build build -j $(nproc)
```
And The error is : **c++: fatal error: Killed signal terminated program cc1plus** 

I've  downloaded **build** file from https://drive.google.com/drive/folders/1Ah580EIylJJ0v2vGOeSBU_b8zPDWESxS ,and unzip it on the  genesis/ext/LuisaRender , i've also checked there has a **LuisaRenderPy.cpython-39-x86_64-linux-gnu.so** on the bin path.
When run examples/rendering/demo.py,get error **No module named 'LuisaRenderPy'**
```python
(torch) root@zhangiser-pc:/usr/Genesis# /root/miniconda3/envs/torch/bin/python /usr/Genesis/examples/rendering/demo.py
[Genesis] [15:43:44] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [15:43:44] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [15:43:44] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [15:43:46] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[Genesis] [15:43:47] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.4, commit 0131dce9, linux, python 3.9.21
[Genesis] [15:43:47] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [15:43:47] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/vis/raytracer.py", line 17, in <module>
    import LuisaRenderPy
ModuleNotFoundError: No module named 'LuisaRenderPy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/Genesis/examples/rendering/demo.py", line 168, in <module>
    main()
  File "/usr/Genesis/examples/rendering/demo.py", line 13, in main
    scene = gs.Scene(
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/vis/visualizer.py", line 60, in __init__
    from .raytracer import Raytracer
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/vis/raytracer.py", line 19, in <module>
    gs.raise_exception(f"Failed to import LuisaRenderer. {e.__class__.__name__}: {e}")

[Genesis] [15:43:49] [ERROR] Failed to import LuisaRenderer. ModuleNotFoundError: No module named 'LuisaRenderPy'
```

Genozen, 2024-12-29:
You have to keep building the LuisaRender until it's successful via `cmake --build build -j $(nproc)` . If your computer freezes alot, tyr using 1 or 2 cores `-j 1`. My PC froze couple times, but the built progress adds up until 100% built success.

zhangiser, 2024-12-29:
> You have to keep building the LuisaRender until it's successful via `cmake --build build -j $(nproc)` . If your computer freezes alot, tyr using 1 or 2 cores `-j 1`. My PC froze couple times, but the built progress adds up until 100% built success.

Thanks，I tried a few times，at last get 100% built success。
[image]

But still get error **ModuleNotFoundError: No module named 'LuisaRenderPy'**
```python
(torch) root@zhangiser-pc:/usr/Genesis# /root/miniconda3/envs/torch/bin/python /usr/Genesis/examples/rendering/demo.py
[Genesis] [17:25:55] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [17:25:55] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [17:25:55] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [17:25:59] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[Genesis] [17:26:00] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.4, commit 0131dce9, linux, python 3.9.21
[Genesis] [17:26:00] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [17:26:00] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/vis/raytracer.py", line 17, in <module>
    import LuisaRenderPy
ModuleNotFoundError: No module named 'LuisaRenderPy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/Genesis/examples/rendering/demo.py", line 168, in <module>
    main()
  File "/usr/Genesis/examples/rendering/demo.py", line 13, in main
    scene = gs.Scene(
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/vis/visualizer.py", line 60, in __init__
    from .raytracer import Raytracer
  File "/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/vis/raytracer.py", line 19, in <module>
    gs.raise_exception(f"Failed to import LuisaRenderer. {e.__class__.__name__}: {e}")

[Genesis] [17:26:02] [ERROR] Failed to import LuisaRenderer. ModuleNotFoundError: No module named 'LuisaRenderPy'
[Genesis] [17:26:02] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

zhouxian, 2024-12-29:
Yeah sometimes multiprocessing building could be problematic

zhangiser, 2024-12-29:
solved

closed 2024-12-29 by zhangiser