Genesis-Embodied-AI/genesis-world #386: examples/rendering/demo can see nothing

zhangiser, 2024-12-29:
I just installed LuisaRenderPy,and run examples/rendering/demo,there show a black window,but i can see nothing
```python
(torch) root@zhangiser-pc:/usr/Genesis# /root/miniconda3/envs/torch/bin/python /usr/Genesis/examples/rendering/demo.py
[Genesis] [22:17:18] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [22:17:18] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [22:17:18] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [22:17:20] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[Genesis] [22:17:20] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.4, commit 0131dce9, linux, python 3.9.21
[Genesis] [22:17:20] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [22:17:20] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [22:17:24] [INFO] Scene <358c738> created.
[Genesis] [22:17:24] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <5fa03d3>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [22:17:24] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 0.
[Genesis] [22:17:24] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <da43849>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 1.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <85c3f8e>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 2.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 3, uid: <b61704d>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 3.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 4, uid: <acc18d5>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 4.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 5, uid: <adbd61d>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 5.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 6, uid: <6968ca3>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 6.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 7, uid: <dc6c019>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:25] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 7.
[Genesis] [22:17:25] [INFO] Adding <gs.RigidEntity>. idx: 8, uid: <309d8a4>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/wooden_sphere_OBJ/wooden_sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:26] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 8.
[Genesis] [22:17:26] [INFO] Adding <gs.RigidEntity>. idx: 9, uid: <5a9c52b>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/wooden_sphere_OBJ/wooden_sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [22:17:27] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 9.
[Genesis] [22:17:27] [INFO] Building scene <358c738>...
[Genesis] [22:17:30] [INFO] Compiling simulation kernels...
[Genesis] [22:17:33] [INFO] Building visualizer...
Segmentation fault (core dumped)
```
[image]

zhouxian, 2024-12-29:
could you provide more info? os version, wsl or native etc

is this a freshly installed conda env?

coghks625, 2024-12-30:
https://github.com/Genesis-Embodied-AI/Genesis/issues/318
i have same problem

zhangiser, 2024-12-30:
> could you provide more info? os version, wsl or native etc
> 
> is this a freshly installed conda env?

I'm used windows wsl ubuntu,and this is a new conda env
version info:
```bash
WSL:2
Ubuntu:22.04
CUDA:12.4
conda:24.11.1
Pytorch:2.5.1
Genesis：main
```

nvcc info:
```bash
(base) root@zhangiser-pc:~# nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2024 NVIDIA Corporation
Built on Thu_Mar_28_02:18:24_PDT_2024
Cuda compilation tools, release 12.4, V12.4.131
Build cuda_12.4.r12.4/compiler.34097967_0
```
nvidia-smi info:
```bash
(base) root@zhangiser-pc:~# nvidia-smi
Mon Dec 30 10:09:49 2024
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 565.77.01              Driver Version: 566.36         CUDA Version: 12.7     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4080 ...    On  |   00000000:01:00.0 Off |                  N/A |
| N/A   46C    P0            593W /  155W |       0MiB /  12282MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```
pytorch info:
```bash
(base) root@zhangiser-pc:~# conda activate torch
(torch) root@zhangiser-pc:~# python
Python 3.9.21 | packaged by conda-forge | (main, Dec  5 2024, 13:51:40)
[GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> torch.cuda.is_available()
True
```
demo.py
```python
import os
os.environ['PYOPENGL_PLATFORM'] = 'glx'
import torch
import importlib.util
import genesis as gs

def main():
    ########################## init ##########################
    gs.init(seed=0, precision="32", logging_level="debug")
[... 134 lines cut ...]
        ),
    )
    ########################## cameras ##########################
    cam_0 = scene.add_camera(
        res=(1600, 900),
        pos=(8.5, 0.0, 1.5),
        lookat=(3.0, 0.0, 0.7),
        fov=60,
        GUI=True,
        spp=512,
    )
    scene.build()

    ########################## forward + backward twice ##########################
    scene.reset()
    horizon = 2000

    for i in range(horizon):
        scene.step()
        cam_0.render()

if __name__ == "__main__":
    main()

```

Kashu7100, 2024-12-30:
Can you try to install nvidia-driver-550 for CUDA12.4?

coghks625, 2024-12-30:
It seems that there are still issues with CUDA 12.4. After various attempts, I confirmed that it works without errors in the CUDA 12.1 environment.

zhangiser, 2024-12-30:
@Kashu7100 @coghks625 
I have tried download NVIDIA driver 550,but only found 552.44 from[ nvdia offical website History Version](https://www.nvidia.cn/drivers/results/).

[image]

And the error is still there.
nvidia-smi info:
```bash
C:\Users\zhangiser>nvidia-smi
Mon Dec 30 20:10:27 2024
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 552.44                 Driver Version: 552.44         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                     TCC/WDDM  | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4080 ...  WDDM  |   00000000:01:00.0 Off |                  N/A |
| N/A   45C    P0             29W /  155W |       0MiB /  12282MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```
error info[**Segmentation fault (core dumped)**]:
(torch) root@zhangiser-pc:/usr/Genesis# /root/miniconda3/envs/torch/bin/python /usr/Genesis/examples/rendering/demo.py
[Genesis] [18:53:33] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [18:53:33] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [18:53:33] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [18:53:34] [INFO] Running on [NVIDIA GeForce RTX 4080 Laptop GPU] with backend gs.cuda. Device memory: 11.99 GB.
[Genesis] [18:53:36] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.4, commit 0131dce9, linux, python 3.9.21
[Genesis] [18:53:36] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [18:53:36] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [18:53:39] [INFO] Scene <579af45> created.
[Genesis] [18:53:39] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <c233424>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [18:53:39] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 0.
[Genesis] [18:53:39] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <ddf49f8>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 1.
[Genesis] [18:53:40] [INFO] Adding <gs.RigidEntity>. idx: 2, uid: <5284270>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 2.
[Genesis] [18:53:40] [INFO] Adding <gs.RigidEntity>. idx: 3, uid: <83c3566>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 3.
[Genesis] [18:53:40] [INFO] Adding <gs.RigidEntity>. idx: 4, uid: <47933bc>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 4.
[Genesis] [18:53:40] [INFO] Adding <gs.RigidEntity>. idx: 5, uid: <849b29b>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 5.
[Genesis] [18:53:40] [INFO] Adding <gs.RigidEntity>. idx: 6, uid: <8b3d6fc>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 6.
[Genesis] [18:53:40] [INFO] Adding <gs.RigidEntity>. idx: 7, uid: <beb84a4>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:40] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 7.
[... cut ...]
[Genesis] [18:53:41] [INFO] Adding <gs.RigidEntity>. idx: 9, uid: <8dde267>, morph: <gs.morphs.Mesh(file='/root/miniconda3/envs/torch/lib/python3.9/site-packages/genesis/assets/meshes/wooden_sphere_OBJ/wooden_sphere.obj')>, material: <gs.materials.Rigid>.
[Genesis] [18:53:42] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 9.
[Genesis] [18:53:42] [INFO] Building scene <579af45>...
[Genesis] [18:53:45] [INFO] Compiling simulation kernels...
[Genesis] [18:53:48] [INFO] Building visualizer...
Segmentation fault (core dumped)

coghks625, 2024-12-31:
@zhangiser try cuda 12.1

zhangiser, 2024-12-31:
@coghks625 

I have tried cuda12.1 but also get Segmentation fault (core dumped)
nvcc info:
```bash
(base) root@zhangiser-pc:/usr/wsl# nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Mon_Apr__3_17:16:06_PDT_2023
Cuda compilation tools, release 12.1, V12.1.105
Build cuda_12.1.r12.1/compiler.32688072_0
```
nvidia-smi:
```bash
(base) root@zhangiser-pc:/usr/wsl# nvidia-smi
Tue Dec 31 10:50:15 2024
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 550.76.01              Driver Version: 552.44         CUDA Version: 12.4     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4080 ...    On  |   00000000:01:00.0 Off |                  N/A |
| N/A   48C    P0             26W /  155W |       0MiB /  12282MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|  No running processes found                                                             |
+-----------------------------------------------------------------------------------------+
```
torch info:
```bash
(base) root@zhangiser-pc:/usr/wsl# conda activate torch
(torch) root@zhangiser-pc:/usr/wsl# python
Python 3.9.21 | packaged by conda-forge | (main, Dec  5 2024, 13:51:40)
[GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> torch.cuda.is_available()
True
```
Guess maybe other problems.
I have tried nvidia driver 566 522 and cuda 12.4.2  12.4.1 12.1.1 until now。

zhangiser, 2025-01-03:
Hello Genesis also get Segmentation fault (core dumped) at current env
```bash
WSL:2
CUDA:12.1
Ubuntu:22.04
conda:24.11.1
Pytorch:2.5.1
NVIDIA Driver：552.44
Python：3.9.21
```
But Hello Genesis run success at old env
```bash
WSL:2
CUDA:12.4.2
Ubuntu:22.04
conda:24.11.1
Pytorch:2.5.1
NVIDIA Driver：566.25
Python：3.9.21
```

Kashu7100, 2025-01-03:
Can you try nvidia driver 535 for 12.1?

zhangiser, 2025-01-04:
> Can you try nvidia driver 535 for 12.1?

Nope,why the version you suggest is so old? The oldest  version i can find from nvidia office website is 552.And it's not worked for me.

myamazum, 2025-01-05:
If your pc has two graphic cards, following comment may help your problem, I guess.

https://github.com/Genesis-Embodied-AI/Genesis/issues/102#issuecomment-2571555626

zhangiser, 2025-01-05:
Now i confirm the problem `Segmentation fault (core dumped)` caused by LuisaRender.
I created a new conda env after back to my origin env:
```bash
WSL:2
CUDA:12.4.2
Ubuntu:22.04
conda:24.11.1
Pytorch:2.5.1
NVIDIA Driver：566.36
Python：3.9.21
```
Now i run tutorials/Hello Genesis success ,  tutorials/visualization success.
[image]
[image]

But After I build LuisaRender and copied LuisaRender to conda env `site_packages/genesis/ext/LuisaRender`. I run Hello_Genesis.py and visualization.py again, i will get the error `Segmentation fault (core dumped)`.

At first, I thought there was a problem with my building LuisaRender, but when I copied the [official LuisaRender  build](https://drive.google.com/drive/folders/1Ah580EIylJJ0v2vGOeSBU_b8zPDWESxS) to `site_packages/genesis/ext/LuisaRender`, i found the `Segmentation fault (core dumped)` has not changed.

And i found that if you nvidia driver changed,it's best to recreate the conda environment and install genesis to run Hello Genesis,as the success rate is relatively high.

Kashu7100, 2025-01-05:
I am not sure what you mean by official LuisaRender build but Genesis is using its own variant of Luisa (under genesis/ext) so if you use the original Luisa, that will be problematic.

Kashu7100, 2025-01-06:
Can you try compiling on your machine?

zhangiser, 2025-01-06:
@Kashu7100 
Yep, i have tried compile on my machine,but failed,see [486](https://github.com/Genesis-Embodied-AI/Genesis/issues/486)

YilingQiao, 2025-03-06:
#465 and #486 mentioned a similar issue which is resolved by
```
This issue can be solved by downgrading the driver to a version compatible to CUDA (12.1 - 535, 12.4 - 550 according to what mentioned in the answer https://github.com/Genesis-Embodied-AI/Genesis/issues/465#issuecomment-2570018322. I tried the 12.1 - 535 one.
```