Genesis-Embodied-AI/genesis-world #391: URDF Model Fails to Stay on Plane and Gravity Customization Issues

Richal13Yu, 2024-12-29:
When I load my custom URDF model into Genesis, the model consistently 'falls' and fails to remain stationary on the plane. Additionally, I am unable to customize the gravity settings for the URDF model. I have also observed that the gs.material functionality is not usable.

- The URDF model should remain stationary on the plane upon loading.
-  Gravity settings for the URDF should be customizable.
-  gs.material should function as expected.

[image]
[image]

Richal13Yu, 2024-12-29:
this code will be fine with my prob
psm = scene.add_entity(
    gs.morphs.URDF(file='urdf/psm/psm_RL.urdf',
                   fixed=True,
                   collision=True,
                   pos=(1.0, 1.0, 0.0)
                   ),

zhouxian, 2024-12-30:
What do you mean it’s not usable?

zhouxian, 2024-12-30:
falling apart is expected behavior, if gravity compensation is turned off and you are not sending any control signal

zhouxian, 2024-12-30:
Ok i think i misunderstood - could you show a video of "falling apart"? if the links are connected via joints, they should be connected even under gravity.

gravity compensation can be specified when you create entity: `add_entity(material=gs.materials.Rigid(gravity_compensation=1.0)`

you can check the robot's links and joints by simply typing `robot.joints` or `robot.links` in a interactive shell

Richal13Yu, 2024-12-30:
### 

> Ok i think i misunderstood - could you show a video of "falling apart"? if the links are connected via joints, they should be connected even under gravity.
>
> gravity compensation can be specified [...]

Thank you for the clarification. In my current URDF file, the links are already connected via joints as far as I can tell. However, the robot still falls apart in this case. Could you please help me identify what might be causing this issue?
[psm.webm](https://github.com/user-attachments/assets/2a89d31a-5021-4a88-87de-6693d4a2154f)

Kashu7100, 2024-12-31:
Could you try `set_dofs_position()` to your robot's dof? I think your robot is rotating due to the gravity.

closed 2025-03-06 by YilingQiao