Genesis-Embodied-AI/genesis-world #392: GPU device not detected inside docker

anahrendra, 2024-12-30:
**Description**
I tried installing the docker environment to run Genesis, but sadly the GPUs are not detected via `torch.cuda.is_available()` inside docker.

`nvidia-smi` works fine and I can see my GPUs there. Also, the initial version of Genesis installed via pip works on my local conda environment (GPUs were detected). Any ideas on what may be causing the issues?

Note: Genesis is installed without any issue in Docker, I can run other CPU-based examples already.

**Environment:**
OS: Ubuntu 18.04
GPUs: 1 RTX3080Ti, 1 RTX3060Ti
NVIDIA Driver: 525.105.17
CUDA: 12.0

Thanks for your help!

Kashu7100, 2024-12-30:
I think your host nvidia driver/CUDA version is too old. The Docker image uses 12.1 or 12.4 (or you could create one for 12.0), and considering the forward compatibility, you might want to update your host machine.

dontpanic123, 2025-01-09:
I have the same issue, somebody has any update?

anahrendra, 2025-01-09:
> I have the same issue, somebody has any update?

I think @Kashu7100 is the probable cause for me. But I haven’t check further by reinstalling my desktop OS. 

What is your nvidia driver btw?

closed 2025-02-28 by Kashu7100