Genesis-Embodied-AI/genesis-world #397: "/bin/sh: 1: ./build_luisa.sh: not found" while trying to build the Docker image

git523, 2024-12-30:
I was trying to build the Docker image, and came across the following error. Not sure if anyone has this before, thanks for your help: 

D:\Genesis-docker\github>docker build -t genesis -f docker/Dockerfile docker
[+] Building 819.7s (19/26)                                                                                                             docker:desktop-linux
 => [internal] load build definition from Dockerfile                                                                                                    0.0s
 => => transferring dockerfile: 4.50kB                                                                                                                  0.0s
 => [internal] load metadata for docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel                                                                  0.4s
 => [internal] load .dockerignore                                                                                                                       0.0s
 => => transferring context: 2B                                                                                                                         0.0s
 => [internal] load build context                                                                                                                       0.0s
 => => transferring context: 1.75kB                                                                                                                     0.0s
 => [builder  1/11] FROM docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel@sha256:e8e63dd7baca894ba11fe1ba48a52a550793c8974f89b533d697784dd20a4dc0  0.0s
 => => resolve docker.io/pytorch/pytorch:2.5.1-cuda12.1-cudnn9-devel@sha256:e8e63dd7baca894ba11fe1ba48a52a550793c8974f89b533d697784dd20a4dc0            0.0s
 => CACHED [stage-1  2/12] RUN apt-get update && apt-get install -y --no-install-recommends     tmux     git     curl     wget     bash-completion      0.0s
 => [stage-1  3/12] WORKDIR /workspace                                                                                                                  0.0s
 => CACHED [builder  2/11] RUN apt-get update && apt-get install -y --no-install-recommends     build-essential     manpages-dev     libvulkan-dev      0.0s
 => CACHED [builder  3/11] RUN add-apt-repository ppa:ubuntu-toolchain-r/test &&     apt update &&     apt install -y --no-install-recommends     gcc-  0.0s
 => CACHED [builder  4/11] RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 110 &&     update-alternatives --install /usr/bin/g++ g+  0.0s
 => [builder  5/11] RUN curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y                                                        77.4s
 => [stage-1  4/12] RUN pip install --no-cache-dir open3d                                                                                             409.0s
 => [builder  6/11] RUN pip install "pybind11[global]"                                                                                                  2.4s
 => [builder  7/11] RUN wget https://github.com/Kitware/CMake/releases/download/v3.31.0-rc2/cmake-3.31.0-rc2-linux-x86_64.sh &&     chmod +x cmake-3.  32.8s
 => [builder  8/11] WORKDIR /workspace                                                                                                                  0.0s
 => [builder  9/11] RUN git clone https://github.com/Genesis-Embodied-AI/Genesis.git &&     cd Genesis &&     git submodule update --init --recursiv  706.2s
 => CANCELED [stage-1  5/12] RUN git clone https://github.com/Genesis-Embodied-AI/Genesis.git &&     cd Genesis &&     pip install . &&     pip inst  410.1s
 => [builder 10/11] COPY build_luisa.sh /workspace/build_luisa.sh                                                                                       0.1s
 => ERROR [builder 11/11] RUN chmod +x ./build_luisa.sh && ./build_luisa.sh 3.11                                                                        0.3s
------
 > [builder 11/11] RUN chmod +x ./build_luisa.sh && ./build_luisa.sh 3.11:
0.238 /bin/sh: 1: ./build_luisa.sh: not found
------
Dockerfile:52
--------------------
  50 |         git submodule update --init --recursive
  51 |     COPY build_luisa.sh /workspace/build_luisa.sh
  52 | >>> RUN chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}
  53 |
  54 |     # ===============================================================
--------------------
ERROR: failed to solve: process "/bin/sh -c chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}" did not complete successfully: exit code: 127

AgentOfLyra, 2024-12-30:
我也遇到了，似乎是clone luisa的时候出现了问题导致clone失败

AgentOfLyra, 2024-12-30:
我在windows中构建的时候也遇到了这个问题

alanahmet, 2024-12-30:
This usually happens on Windows because the shell script has DOS/Windows CRLF line endings rather than Unix LF line endings. Even though you see the file in ls -l and have executed chmod +x, the Docker build environment (which is Linux-based) treats the carriage return (\r) as part of the filename or a bad interpreter directive, causing the not found error.

[image]

You can fix it by changing CRLF line editing to LF line editing on VSCode

git523, 2025-01-01:
Thank you very much, I made the change of CRLF to LF, and the image-building process did proceed, and it came across another issue later down the path: 

=================
33.81 [ 25%] Linking C shared library ../../../../../bin/liblc-ext-stb.so
33.82 [ 25%] Built target stb
33.83 [ 25%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/stl/filesystem.cpp.o
33.83 [ 25%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/basic_types.cpp.o
33.83 [ 25%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/stl/memory.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/binary_buffer.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/stl/hash.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/binary_file_stream.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/binary_io.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/dynamic_module.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/first_fit.cpp.o
33.83 [ 26%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/platform.cpp.o
33.83 [ 27%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/pool.cpp.o
33.83 [ 27%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/logging.cpp.o
33.83 [ 27%] Building CXX object src/compute/src/core/CMakeFiles/luisa-compute-core.dir/string_scratch.cpp.o
34.10    Compiling serde_derive v1.0.193
34.10    Compiling zerocopy-derive v0.6.6
34.43    Compiling zerocopy v0.6.6
35.38 gmake[1]: *** [CMakeFiles/Makefile2:1291: src/ext/assimp/code/CMakeFiles/assimp.dir/all] Error 2
35.38 gmake[1]: *** Waiting for unfinished jobs....
36.04    Compiling embree_sys v0.1.11 (https://github.com/shiinamiyuki/embree-sys#6a0a591d)
36.30 /workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/core/binary_file_stream.cpp: In member function 'virtual void luisa::BinaryFileStream::read(eastl::span<std::byte>)':
36.30 /workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/core/binary_file_stream.cpp:68:10: warning: ignoring return value of 'size_t fread(void*, size_t, size_t, FILE*)' declared with attribute 'warn_unused_result' [-Wunused-result]
36.30    68 |     fread(dst.data(), 1, size, _file);
36.30       |     ~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
36.37    Compiling toml v0.5.11
36.50    Compiling half v2.3.1
36.50    Compiling bincode v1.3.3
38.00    Compiling luisa_compute_api_types v0.1.1-alpha.1 (/workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_api_types)
38.00    Compiling luisa_compute_ir v0.1.1-alpha.1 (/workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_ir)
38.00    Compiling luisa_compute_cpu_kernel_defs v0.1.0-alpha.1 (/workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_cpu_kernel_defs)
39.83 [ 27%] Linking CXX shared library ../../../../bin/liblc-core.so
39.88 [ 27%] Built target luisa-compute-core
56.59    Compiling luisa_compute_ir_v2 v0.1.0 (/workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_ir_v2)
56.59    Compiling luisa_compute_backend v0.1.1-alpha.1 (/workspace/Genesis/genesis/ext/LuisaRender/src/compute/src/rust/luisa_compute_backend)
56.63 warning: unused import: `crate::*`
56.63  --> luisa_compute_ir_v2/src/convert.rs:1:5
56.63   |
56.63 1 | use crate::*;
56.63   |     ^^^^^^^^
56.63   |
56.63   = note: `#[warn(unused_imports)]` on by default
56.63
56.74 warning: unused variable: `module`
56.74   --> luisa_compute_ir_v2/src/convert.rs:10:20
56.74    |
56.74 10 |     pub fn convert(module:ir_v1::ir::Module) {
56.74    |                    ^^^^^^ help: if this is intentional, prefix it with an underscore: `_module`
56.74    |
56.74    = note: `#[warn(unused_variables)]` on by default
56.74
56.75 warning: struct `V1ToV2` is never constructed
56.75  --> luisa_compute_ir_v2/src/convert.rs:5:8
56.75   |
56.75 5 | struct V1ToV2 {
56.75   |        ^^^^^^
56.75   |
56.75   = note: `#[warn(dead_code)]` on by default
56.75
56.75 warning: associated function `convert` is never used
56.75   --> luisa_compute_ir_v2/src/convert.rs:10:12
56.75    |
56.75 9  | impl V1ToV2 {
56.75    | ----------- associated function in this implementation
56.75 10 |     pub fn convert(module:ir_v1::ir::Module) {
56.75    |            ^^^^^^^
56.75
[... 1141 lines cut ...]
87.82    --> luisa_compute_backend_impl/src/cpu/texture.rs:243:19
87.82     |
87.82 142 | impl TextureView {
87.82     | ---------------- method in this implementation
87.82 ...
87.82 243 |     pub(crate) fn copy_to_vec_2d(&self) -> Vec<u8> {
87.82     |                   ^^^^^^^^^^^^^^
87.82
87.82 warning: `luisa_compute_backend_impl` (lib) generated 28 warnings (run `cargo fix --lib -p luisa_compute_backend_impl` to apply 7 suggestions)
87.82     Finished `release` profile [optimized] target(s) in 1m 00s
87.83 [ 28%] Copying embree to the output directory...
87.83 [ 28%] Generating ../../../../bin/libluisa_compute_api_types.a
87.85 [ 28%] Built target libluisa_compute_api_types.a
87.88 [ 28%] Built target luisa_compute_rust_build_embree
87.88 gmake: *** [Makefile:136: all] Error 2
------
Dockerfile:52
--------------------
  50 |         git submodule update --init --recursive
  51 |     COPY build_luisa.sh /workspace/build_luisa.sh
  52 | >>> RUN chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}
  53 |
  54 |     # ===============================================================
--------------------
ERROR: failed to solve: process "/bin/sh -c chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}" did not complete successfully: exit code: 2

SphaeroX, 2025-01-01:
Switched from CRFL to LF and it works. thank @alanahmet 

@git523 open the file, switch to LF and than save the file. Run the build again.

git523, 2025-01-01:
Thank you @SphaeroX , I tried again and still could not move forward, each time the error message differs a little though. I am not sure if it could be related to the downloading process(?)....

=========
**(i only copy/paste part of the log file here)

106.2 1295 |     fn gen_instr(&mut self, node: NodeRef) {
106.2      |        ^^^^^^^^^
106.2 ...
106.2 1531 |     fn gen_block_(&mut self, block: BasicBlockRef) {
106.2      |        ^^^^^^^^^^
106.2 ...
106.2 1550 |     fn gen_block(&mut self, block: BasicBlockRef) {
106.2      |        ^^^^^^^^^
106.2 ...
106.2 1559 |     fn gen_callable(
106.2      |        ^^^^^^^^^^^^
106.2
106.2 warning: method `copy_to_vec_2d` is never used
106.2    --> luisa_compute_backend_impl/src/cpu/texture.rs:243:19
106.2     |
106.2 142 | impl TextureView {
106.2     | ---------------- method in this implementation
106.2 ...
106.2 243 |     pub(crate) fn copy_to_vec_2d(&self) -> Vec<u8> {
106.2     |                   ^^^^^^^^^^^^^^
106.2
107.1 warning: `luisa_compute_backend_impl` (lib) generated 28 warnings (run `cargo fix --lib -p luisa_compute_backend_impl` to apply 7 suggestions)
107.1     Finished `release` profile [optimized] target(s) in 1m 28s
107.1 [ 49%] Copying /workspace/Genesis/genesis/ext/LuisaRender/build/src/compute/src/rust/rust/release/libluisa_compute_backend_impl.so to /workspace/Genesis/genesis/ext/LuisaRender/build/bin/libluisa_compute_backend_impl.so and patch it
107.1 [ 49%] Built target luisa_compute_rust_build_libluisa_compute_backend_impl.so
107.1 [ 49%] Built target luisa_compute_rust_build
107.1 [ 49%] Building CXX object src/compute/src/ir/CMakeFiles/luisa-compute-ir.dir/ast2ir.cpp.o
107.1 [ 49%] Building CXX object src/compute/src/ir/CMakeFiles/luisa-compute-ir.dir/ir2ast.cpp.o
107.1 [ 49%] Building CXX object src/compute/src/ir/CMakeFiles/luisa-compute-ir.dir/transform.cpp.o
115.1 [ 50%] Linking CXX shared library ../../../../bin/liblc-ir.so
115.2 [ 50%] Built target luisa-compute-ir
115.2 [ 50%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/cmake_pch.hxx.gch
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/bindless_array.cpp.o
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/buffer.cpp.o
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/byte_buffer.cpp.o
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/command_list.cpp.o
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/context.cpp.o
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/device.cpp.o
118.8 [ 51%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/dispatch_buffer.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/event.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/image.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/sparse_buffer.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/mipmap.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/volume.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/sparse_heap.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/sparse_texture.cpp.o
118.8 [ 52%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/sparse_command_list.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/stream.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rhi/command_encoder.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rhi/device_interface.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rhi/pixel.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/swapchain.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rhi/resource.cpp.o
118.8 [ 53%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rtx/mesh.cpp.o
118.8 [ 54%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rtx/accel.cpp.o
118.8 [ 54%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/raster/depth_buffer.cpp.o
118.8 [ 54%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rtx/curve.cpp.o
118.8 [ 54%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/raster/raster.cpp.o
118.8 [ 54%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/rtx/procedural_primitive.cpp.o
118.8 [ 54%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/remote/server_interface.cpp.o
118.8 [ 55%] Building CXX object src/compute/src/runtime/CMakeFiles/luisa-compute-runtime.dir/remote/client_interface.cpp.o
124.1 during RTL pass: ira
[... cut ...]
126.6 gmake: *** [Makefile:136: all] Error 2
------
Dockerfile:52
--------------------
  50 |         git submodule update --init --recursive
  51 |     COPY build_luisa.sh /workspace/build_luisa.sh
  52 | >>> RUN chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}
  53 |
  54 |     # ===============================================================
--------------------
ERROR: failed to solve: process "/bin/sh -c chmod +x ./build_luisa.sh && ./build_luisa.sh ${PYTHON_VERSION}" did not complete successfully: exit code: 2**

git523, 2025-01-02:
Just a quick follow-up: I gave up on installing it into the Ubuntu environment through Microsoft App store, and switched to virtualbox and installed a fresh new Ubuntu environment. I was able to install the program with no issues there. (I did came across other issues when running the examples, and i will start a new thread). Thanks again to @SphaeroX and @alanahmet

closed 2025-01-02 by git523