Genesis-Embodied-AI/genesis-world #400: CUDA libraries cannot be found and linked

huang-xiansheng, 2024-12-30:
Two issues occurred whenever the hello,genesis.py is running with backend=gs.gpu.

the first is  [cuda_driver.cpp:load_lib@36] libcuda.so lib not found.
the other is [misc.py:adaptive_arch_select@758] Arch=[<Arch.cuda: 3>] is not supported, falling back to CPU.

I have installed the libcuda.so and set the environment variables but the python still cannot find the correct path.

How can I solve these two problems?

And by the way , if I do not want to receive so much running information such as [INFO] Running at 65.80 FPS , what I should do to disable this output ?

aNoWhale, 2024-12-31:
I have no idea which operation is useful. But I dealt with that. I ran nvcc --version and found no cuda tool kit and then I installed it, and then I found no libcuda.so in /usr/lib/x86_64_linux_gnu, so I made a link to libcuda.so in  /usr/wsl/lib which linked to my cuda for windows. So far, the problem hasn't been dealt. Next day, I restarted my computer and just run python script in bash (not pycharm). The problem disappeared. Wish that do some helps to you.

huang-xiansheng, 2024-12-31:
> I have no idea which operation is useful. But I dealt with that. I ran nvcc --version and found no cuda tool kit and then I installed it, and then I found no libcuda.so in /usr/lib/x86_64_linux_gnu, [...]

I examined my own environment and found the same issues you mentioned , then I also linked my libcuda.so to /usr/lib/x86_64-linux-gnu via a soft link. 

Magically these two issues disappeared immediately and I think these may be caused by what the Genesis calls CUDA via libcuda.so and links to /usr/wsl/lib path where the .so does not exist . 

Thanks for your solution .

closed 2024-12-31 by huang-xiansheng