Genesis-Embodied-AI/genesis-world #401: Problem about tutorial: Inverse Kinematics & Motion Planning

zhengbi-yong, 2024-12-30:
```zsh
[Genesis] [19:37:35] [INFO] Building scene <64f265b>...
[Genesis] [19:37:38] [INFO] Compiling simulation kernels...
[Genesis] [19:37:41] [INFO] Building visualizer...
[Genesis] [19:37:42] [INFO] Viewer created. Resolution: 1440×1080, max_FPS: 60.
[Genesis] [19:37:53] [WARNING] `qpos_start` exceeds joint limit. Relaxing joint limit to contain `qpos_start` for planning.
[1]    1792289 segmentation fault (core dumped)  python IK_motion_planning_grasp.py
```
I tried the `examples/tutorials/IK_motion_planning_grasp.py` but encountered the error above, how can I solve it?

zhouxian, 2024-12-30:
No this is a warning not an error -- you crash is probably caused by ompl itself :(

zhengbi-yong, 2025-01-02:
> No this is a warning not an error -- you crash is probably caused by ompl itself :(
How to solve this segmentation fault? Reinstall ompl?

longjie0723, 2025-01-05:
I got same error using OMPL of pip install (using the wheel file). [Installing OMPL from source code](https://ompl.kavrakilab.org/core/installation.html) have solved it.

zhengbi-yong, 2025-01-09:
> I got same error using OMPL of pip install (using the wheel file). [Installing OMPL from source code](https://ompl.kavrakilab.org/core/installation.html) have solved it.

Can you please help me in detail? I've tried python3.9-3.12, with the ompl install script provided by https://ompl.kavrakilab.org/core/install-ompl-ubuntu.sh ,but it still not work.

closed 2025-03-24 by duburcqa