Genesis-Embodied-AI/genesis-world #402: Feature suggestion. Interface to draw debug information.

Riften, 2024-12-30:
I hope more drawing interfaces can be provided to facilitate scene debugging. For example, interfaces for drawing various primitive shapes.

I notice that the default viewer is implemented through pyrender. I successfully draw some primitive shapes through pyrender's drawing interface, but I am not sure if this will affect the simulation process or the rendering of the camera sensor.

https://github.com/user-attachments/assets/b98009fc-593a-4f17-8571-dd4b765f13a5

<details>
<summary>code for drawing</summary>

```python
import genesis as gs
from genesis.ext.pyrender import Scene as PyScene
from genesis.ext.pyrender import Mesh as PyMesh
from genesis.ext.pyrender import Node as PyNode
import genesis.ext.trimesh as trimesh
import numpy as np

gs.init(backend=gs.cpu)

scene = gs.Scene()

plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    # gs.morphs.URDF(
    #     file='urdf/panda_bullet/panda.urdf',
    #     fixed=True,
    # ),
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
)

scene.build()
pyrender_scene = scene.viewer._pyrender_viewer.scene
# create a cube mesh through trimesh

cube = trimesh.creation.box(extents=[1.0, 0.5, 0.5])
cube_mesh = PyMesh.from_trimesh(cube)
cube_node = PyNode(mesh=cube_mesh, translation=np.array([0.1, 0.15, 0]))
pyrender_scene.add_node(cube_node)
for i in range(100):
    scene.step()
```

</details>

Is there any plan to support such feature?

zhouxian, 2024-12-30:
We do have! If you look into `scene.py` we have something like `draw_debug_points` `draw_debug_box` etc

Riften, 2024-12-30:
Thank you for the response! I will try these functions.

closed 2024-12-30 by Riften