Genesis-Embodied-AI/genesis-world #403: Segmentation Fault in test_rigid_island.py

andrewlyu, 2024-12-30:
Running the test_rigid_island.py script in a Python 3.11 Anaconda environment (macOS) results in a segmentation fault. 
The script outputs FPS values during execution but ultimately crashes without completing.

Observed Behavior:
* Warnings: Duplicate class definitions for GLFWHelper, GLFWApplicationDelegate, GLFWWindowDelegate, GLFWContentView, and GLFWWindow.
* FPS values observed in the test output:
True island  | 1 cubes  | <CG: 0>  | 38,318,023.17 fps  | 8192 envs  
False island | 1 cubes  | <CG: 0>  | 36,065,914.07 fps  | 8192 envs 
* The script terminates with a segmentation fault:
zsh: segmentation fault  python test_rigid_island.py

A detailed crash report is attached.
[crash_12_30_2024.txt](https://github.com/user-attachments/files/18275789/crash_12_30_2024.txt)

formoree, 2024-12-31:
I also run this file on my device (4090 Cuda:12.4), I got following error.
```
/home/user/miniconda3/envs/genesis/lib/python3.9/site-packages/collada/schema.py:21: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import resource_string
double free or corruption (out)
Aborted (core dumped)
```

duburcqa, 2025-03-22:
Could you try on main branch ?

closed 2025-05-24 by duburcqa