Genesis-Embodied-AI/genesis-world #408: Genesis Installation on Ubuntu CPU only

hpssjellis, 2024-12-30:
I have a CUDA GPU machine at work but would like to use Genesis on my dual boot windows/Linux laptop (Using Ubuntu from boot not WSL). Does anyone have a link to a non-cuda installation with working CPU only example code? 

My problem is the LuisaRender both viewer and mp4 creation. I can't completely compile it. I will try several more virtual environment pip installs to see what I can do.

I just need a working set of Linux (ubuntu)  installation instructions and 2 or 3 examples for CPU only viewer and MP4 generation. The rest I should be good with.

zhouxian, 2024-12-31:
Luisa requires CUDA to work unfortunately

hpssjellis, 2024-12-31:
@zhouxian After downgrading PyOpenGL to 3.1.5 from 3.1.7 I seem to be getting most of the examples working by watching for the -v argument and changing the backend to cpu. My laptop is a dual boot windows / linux ubuntu with 13th Gen Intel(R) Core(TM) i9-13900H   2.60 GHz and 32 Gb ram.  I am having success on the Ubuntu side.   I will work through all the examples, having issues with the drone, but coupling, tutorial and all the main examples folder ones seem fine.  

I definitely had issues compiling LuisaRenderer, but presently not needing it for basic operation.
 
https://github.com/user-attachments/assets/f41a2fc7-75f2-4ff4-87b9-9c469d2a462b

closed 2024-12-31 by hpssjellis