Genesis-Embodied-AI/genesis-world #413: Request: Add Support for Reading BVH Files

tink4r, 2024-12-31:
I would like to request the addition of functionality to read BVH files. Many users rely on BVH files to import MOCAP data.
Thank you for considering this!

zhouxian, 2024-12-31:
Could you provide a simple example and explain the use case?

tink4r, 2024-12-31:
Use case: BVH files are commonly used in research direction of human animation. Many people use BVH files to read motion capture data to analyze human motion and achieve precise control of character motion in various environments.

This is an example BVH file that contains motion capture data for a human jumping. The file consists of two parts: the **hierarchy** (skeleton) and the **motion** data (position and rotation of joints over time).

> HIERARCHY
ROOT Hips
{
	OFFSET 0.00 0.00 0.00
	CHANNELS 6 Xposition Yposition Zposition Xrotation Yrotation Zrotation
	JOINT LeftUpLeg
	{
		OFFSET 3.64953 0.00000 0.00000
		CHANNELS 3 Xrotation Yrotation Zrotation
		JOINT LeftLeg
		{
			OFFSET 0.00000 -15.70580 0.00000
			CHANNELS 3 Xrotation Yrotation Zrotation
			JOINT LeftFoot
			{
				OFFSET 0.00000 -15.41867 0.00000
				CHANNELS 3 Xrotation Yrotation Zrotation
				JOINT LeftToeBase
				{
					OFFSET 0.00000 -1.53543 5.73033
					CHANNELS 3 Xrotation Yrotation Zrotation
					End Site
					{
						OFFSET 0.00000 0.00000 2.95275
					}
				}
			}
		}
	}
	JOINT RightUpLeg
	{
		OFFSET -3.64953 0.00000 0.00000
		CHANNELS 3 Xrotation Yrotation Zrotation
		JOINT RightLeg
		{
			OFFSET 0.00000 -15.70580 0.00000
			CHANNELS 3 Xrotation Yrotation Zrotation
			JOINT RightFoot
			{
				OFFSET 0.00000 -15.41867 0.00000
				CHANNELS 3 Xrotation Yrotation Zrotation
				JOINT RightToeBase
				{
					OFFSET 0.00000 -1.53543 5.73033
					CHANNELS 3 Xrotation Yrotation Zrotation
					End Site
					{
						OFFSET 0.00000 0.00000 2.95275
					}
				}
			}
		}
	}
	JOINT Spine
	{
		OFFSET 0.00000 0.03937 0.00000
		CHANNELS 3 Xrotation Yrotation Zrotation
		JOINT Spine1
		{
			OFFSET 0.00000 10.24829 0.00000
			CHANNELS 3 Xrotation Yrotation Zrotation
			JOINT Neck
			{
				OFFSET 0.00000 7.82687 0.00000
				CHANNELS 3 Xrotation Yrotation Zrotation
				JOINT Head
				{
					OFFSET 0.00000 6.90715 0.00000
					CHANNELS 3 Xrotation Yrotation Zrotation
					End Site
					{
						OFFSET 0.00000 4.52755 0.00000
					}
				}
			}
			JOINT LeftShoulder
			{
				OFFSET 0.00000 7.82687 0.00000
				CHANNELS 3 Xrotation Yrotation Zrotation
				JOINT LeftArm
				{
					OFFSET 6.71018 -0.00002 0.00000
					CHANNELS 3 Xrotation Yrotation Zrotation
					JOINT LeftForeArm
					{
						OFFSET 10.94419 -0.00004 0.00000
						CHANNELS 3 Xrotation Yrotation Zrotation
						JOINT LeftHand
						{
							OFFSET 8.52010 -0.00003 0.00000
							CHANNELS 3 Xrotation Yrotation Zrotation
							JOINT LeftHandThumb
							{
								OFFSET 0.00000 0.00000 0.00000
								CHANNELS 3 Xrotation Yrotation Zrotation
								End Site
								{
									OFFSET 0.00000 0.00000 3.93700
								}
							}
							JOINT L_Wrist_End
							{
								OFFSET 3.93700 -0.00001 0.00000
								CHANNELS 3 Xrotation Yrotation Zrotation
							}
						}
					}
				}
			}
			JOINT RightShoulder
			{
				OFFSET 0.00000 7.82687 0.00000
				CHANNELS 3 Xrotation Yrotation Zrotation
				JOINT RightArm
				{
					OFFSET -6.71018 -0.00002 0.00000
					CHANNELS 3 Xrotation Yrotation Zrotation
					JOINT RightForeArm
					{
						OFFSET -10.94419 -0.00004 0.00000
						CHANNELS 3 Xrotation Yrotation Zrotation
						JOINT RightHand
						{
							OFFSET -8.52010 -0.00003 0.00000
							CHANNELS 3 Xrotation Yrotation Zrotation
							JOINT RightHandThumb
							{
								OFFSET 0.00000 0.00000 0.00000
								CHANNELS 3 Xrotation Yrotation Zrotation
								End Site
								{
									OFFSET 0.00000 0.00000 3.93700
								}
							}
							JOINT R_Wrist_End
							{
								OFFSET -5.39369 -0.00002 0.00000
								CHANNELS 3 Xrotation Yrotation Zrotation
							}
						}
					}
				}
			}
		}
	}
}
MOTION
Frames: 2575
Frame Time: 0.008333
 1.1473 32.8029 1.7158 1.8673 -8.9395 -2.4851 -1.8048 4.4401 9.1220 10.6891 -0.0000 0.0000 -14.4262 1.6207 -8.1203 -6.6388 0.0000 0.0000 -3.3489 -5.6780 -0.1448 8.2598 -0.0000 0.0000 -19.6835 1.1206 -1.8147 9.2354 0.0000 0.0000 -3.9666 2.1039 -0.9182 11.2827 0.3187 1.3724 3.9228 -10.7439 -1.8312 8.9441 -3.8475 2.9412 18.9681 13.3439 -0.6110 -26.5111 -27.4973 -88.3621 -0.0000 -10.0680 -0.0000 -16.6777 -0.4690 -19.8491 0.0000 0.0000 0.0000 0.0000 0.0000 0.0000 14.7299 -19.0543 5.1655 -22.5208 25.1609 83.9932 -0.0000 9.9763 0.0000 -25.4105 -2.5127 14.0891 -0.0000 -0.0000 0.0000 -0.0000 -0.0000 0.0000
 ...