Genesis-Embodied-AI/genesis-world #414: Why is the fps so low when running the program?

menkeyi001, 2024-12-31:
```shell
(Genesis) root@com1:~/Genesis-main# cat demov1.py
import genesis as gs

gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer = True,
    viewer_options = gs.options.ViewerOptions(
        res           = (1280, 960),
        camera_pos    = (3.5, 0.0, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
        max_FPS       = 60,
    ),
    vis_options = gs.options.VisOptions(
        show_world_frame = True,
        world_frame_size = 1.0,
        show_link_frame  = False,
        show_cameras     = False,
        plane_reflection = True,
        ambient_light    = (0.1, 0.1, 0.1),
    ),
    renderer=gs.renderers.Rasterizer(),
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

cam = scene.add_camera(
    res    = (640, 480),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False,
)

scene.build()

# 渲染rgb、深度、分割掩码和法线图
# rgb, depth, segmentation, normal = cam.render(rgb=True, depth=True, segmentation=True, normal=True)

cam.start_recording()
import numpy as np

for i in range(120):
    scene.step()
    cam.set_pose(
        pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
        lookat = (0, 0, 0.5),
    )
    cam.render()
#cam.stop_recording(save_to_filename='video.mp4', fps=60)

Why is the fps so low when running the program?  Is it a problem with my code that causes the fps to fail? Or hardware resources? I have 4 A100

(Genesis) root@com1:~/Genesis-main# python demov1.py
[Genesis] [07:23:14] [INFO] ╭───────────────────────────────────────────────────────────────────────────╮
[Genesis] [07:23:14] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [07:23:14] [INFO] ╰───────────────────────────────────────────────────────────────────────────╯
[Genesis] [07:23:14] [INFO] Running on [NVIDIA A100 80GB PCIe] with backend gs.cuda. Device memory: 79.15                                                                GB.
[Genesis] [07:23:15] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 d                                                               ebug: False, 🎨 theme: 'dark'.
[Genesis] [07:23:17] [INFO] Scene <91480f5> created.
[Genesis] [07:23:17] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <06976cb>, morph: <gs.morphs.Plane>, ma                                                               terial: <gs.materials.Rigid>.
[Genesis] [07:23:17] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <cf5f819>, morph: <gs.morphs.MJCF(file=                                                               '/root/Genesis-main/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [07:23:18] [INFO] Building scene <91480f5>...
[Genesis] [07:23:23] [INFO] Compiling simulation kernels...
[Genesis] [07:23:30] [INFO] Building visualizer...
[Genesis] [07:23:33] [INFO] Viewer created. Resolution: 1280×960, max_FPS: 60.
[Genesis] [07:23:36] [INFO] Running at 616.38 FPS.
[Genesis] [07:23:43] [INFO] Running at 2.65 FPS.
[Genesis] [07:24:08] [INFO] Running at 0.63 FPS.
[Genesis] [07:24:17] [INFO] Running at 0.51 FPS.
[Genesis] [07:24:36] [INFO] Running at 0.35 FPS.

zhouxian, 2024-12-31:
could you search relevant issues? I think this has been addressed before

ziyanx02, 2025-01-09:
Please refer to #138 for update. I will close this issue because there is no more discussion here.

closed 2025-01-09 by ziyanx02