Genesis-Embodied-AI/genesis-world #415: Visualizer does not work on macOS

gmmyung, 2024-12-31:
Hello, I am trying to test out the visualizer on my m1 MacBook.
I ran the example file: https://github.com/Genesis-Embodied-AI/Genesis/blob/main/examples/render_on_macos.py
And then the code segfaults on its own. 
```
[Genesis] [16:35:13] [INFO] Building scene <6f3066e>...
[Genesis] [16:35:16] [INFO] Compiling simulation kernels...
[Genesis] [16:35:20] [INFO] Building visualizer...
[Genesis] [16:35:21] [WARNING] Non-linux system detected. In order to use the in
teractive viewer, you need to manually run simulation in a separate thread and t
hen start viewer. See `examples/render_on_macos.py`.
[Genesis] [16:35:21] [INFO] Viewer created. Resolution: 600×450, max_FPS: 60.
[Genesis] [16:35:21] [INFO] Running at 311.61 FPS.
15.57506981165706 FPS
[Genesis] [16:35:21] [INFO] Running at 245.45 FPS.
48.48738194051074 FPS
[Genesis] [16:35:21] [INFO] Running at 204.19 FPS.
48.32091796177463 FPS
[Genesis] [16:35:21] [INFO] Running at 176.14 FPS.
48.5907390030005 FPS
[Genesis] [16:35:21] [INFO] Running at 155.77 FPS.
48.32537186178609 FPS
2024-12-31 16:35:21.531 Python[4550:88613] ApplePersistenceIgnoreState: Existing
 state will not be touched. New state will be written to /var/folders/pn/1x_sn6w
s1qz96cvyl5jqjfxh0000gn/T/org.python.python.savedState
[Genesis] [16:35:21] [INFO] Running at 141.35 FPS.
50.76682119125141 FPS
[Genesis] [16:35:21] [INFO] Running at 131.82 FPS.
57.37918958111029 FPS
[Genesis] [16:35:21] [INFO] Running at 121.61 FPS.
48.916018426730425 FPS
[Genesis] [16:35:21] [INFO] Running at 109.48 FPS.
30.55580729527126 FPS
uniform not found: V
uniform not found: P
........

<Repeated Logs>
.........

uniform not found: material.base_color_factor
uniform not found: material.metallic_factor
uniform not found: material.roughness_factor
uniform not found: material.emissive_factor
Traceback (most recent call last):
  File "/Users/gmmyung/Developer/genesis-raibo2/test.py", line 68, in <module>
    main()
  File "/Users/gmmyung/Developer/genesis-raibo2/test.py", line 44, in main
    scene.viewer.start()
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/vis/viewer.py", line 91, in start
    self._pyrender_viewer.start()
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/ext/pyrender/viewer.py", line 386, in start
    self._init_and_start_app()
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._renderer.render_texts(
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/ext/pyrender/renderer.py", line 258, in render_texts
    self._configure_forward_pass_viewport(0)
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/genesis/ext/pyrender/renderer.py", line 1160, in _configure_forward_pass_view
port
    glBindFramebuffer(GL_DRAW_FRAMEBUFFER, 0)
  File "/Users/gmmyung/Developer/genesis-raibo2/.venv/lib/python3.12/site-packag
es/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1282,
        description = b'invalid operation',
        baseOperation = glBindFramebuffer,
        cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)
^C2024-12-31 16:35:37.301 Python[4550:88613] +[IMKClient subclass]: chose IMKCli
ent_Modern
[1]    4550 segmentation fault  python test.py
```

This is my environment details:
- OS: macOS 15.2 (24C101)
- Python: 3.12.8
- PyTorch: 2.5.1
- Genesis: 0.2.0

TristanBester, 2024-12-31:
I ran into the same issue on M1 macbook.

Please help +1

LowPower-Center, 2025-01-01:
same issue on M4 Mac mini 
need help

JeanBon54, 2025-01-06:
I have a Macbook pro M4 and it's works.
I take a lot of time to manage `pip` for python on MacOS.

For this I run:
`python3 render_on_macos.py -V`
But before I did this `pip instal ffmpeg`

gmmyung, 2025-01-06:
What’s your python / PyTorch version btw?

ai-bits, 2025-01-06:
> OS: macOS 15.2 (24C101)
Python: 3.12.8
PyTorch: 2.5.1
Genesis: 0.2.0

Was same here. Ran once I figured -v arg. Viewer set to False by default when the file name suggests the opposite.
Genesis is on 0.2.1 now,.. Did you try to install Genesis from the repo? They've been working over the holidays, though Zhou was the only here to help.
`cd Genesis`
`git pull`
`pip install -U -e .` #from repo
Tried to remove ~/.cache/Genesis?

HTH

duburcqa, 2025-02-25:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/777

closed 2025-02-25 by duburcqa