Genesis-Embodied-AI/genesis-world #426: Error in smoke.py: Unrecognized Attribute res

andrewlyu, 2025-01-01:
Running the smoke.py script fails with an error related to an unrecognized attribute res. The error occurs when initializing SFOptions during the execution of the main() function.

Anaconda, python3.11, m1 macbook

```
Traceback (most recent call last):
  File "/Users/duchaolv/projects/genesis_playground/examples/smoke.py", line 146, in <module>
    main()
  File "/Users/duchaolv/projects/genesis_playground/examples/smoke.py", line 87, in main
    sf_options=gs.options.SFOptions(
               ^^^^^^^^^^^^^^^^^^^^^
  File "/opt/anaconda3/envs/genesis_env/lib/python3.11/site-packages/genesis/options/options.py", line 27, in __init__
    gs.raise_exception(f"Unrecognized attribute: {key}")

[Genesis] [19:30:12] [ERROR] Unrecognized attribute: res
```

Kashu7100, 2025-01-01:
`SFOptions` has `res` attribute. Could you check your Genesis is up to date?

zhrli, 2025-01-03:
> Running the smoke.py script fails with an error related to an unrecognized attribute res. The error occurs when initializing SFOptions during the execution of the main() function.
>
> Anaconda, [...]

Same here in mac3.  It works well in linux.    Have u solved this problem?

zhrli, 2025-01-03:
> Running the smoke.py script fails with an error related to an unrecognized attribute res. The error occurs when initializing SFOptions during the execution of the main() function.
>
> Anaconda, [...]

Solved. Goto python 3.11

closed 2025-03-06 by YilingQiao