Genesis-Embodied-AI/genesis-world #428: Execution example error

zhanglepy, 2025-01-01:
(genesis) zl@zl:~/software/Genesis/examples/tutorials$ python mpm.py -V
[Genesis] [12:07:44] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [12:07:44] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [12:07:44] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [12:07:44] [INFO] Running on [NVIDIA GeForce RTX 4060 Laptop GPU] with backend gs.cuda. Device memory: 7.75 GB.
[Genesis] [12:07:44] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [12:07:44] [INFO] Scene <a8bad34> created.
[Genesis] [12:07:44] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <56276c7>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [12:07:44] [INFO] Adding <gs.MPMEntity>. idx: 1, uid: <8f302c2>, morph: <gs.morphs.Box>, material: <gs.materials.MPM.Elastic>.
[Genesis] [12:07:44] [INFO] Sampled 8,182 particles.
[Genesis] [12:07:44] [INFO] Adding <gs.MPMEntity>. idx: 2, uid: <ae65cfc>, morph: <gs.morphs.Box>, material: <gs.materials.MPM.Liquid>.
[Genesis] [12:07:44] [INFO] Sampled 27,221 particles.
[Genesis] [12:07:44] [INFO] Adding <gs.MPMEntity>. idx: 3, uid: <89360ed>, morph: <gs.morphs.Sphere>, material: <gs.materials.MPM.ElastoPlastic>.
[Genesis] [12:07:44] [INFO] Sampled 4,159 particles.
[Genesis] [12:07:44] [INFO] Building scene <a8bad34>...
[Genesis] [12:07:47] [WARNING] Current `substep_dt` (0.0004) is greater than suggested_dt (0.0003125, calculated based on `grid_density`). Simulation might be unstable.
[Genesis] [12:07:47] [INFO] Compiling simulation kernels...
[Genesis] [12:07:53] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: iris
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/threading.py", line 980, in _bootstrap_inner
    self.run()
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/threading.py", line 917, in run
    self._target(*self._args, **self._kwargs)
  File "/home/zl/software/Genesis/genesis/ext/pyrender/viewer.py", line 1123, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
    super().__init__(*args, **kwargs)
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/site-packages/pyglet/window/__init__.py", line 527, in __init__
    context = config.create_context(gl.current_context)
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 114, in create_context
    return XlibContext(self, share)
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context
^CTraceback (most recent call last):
  File "/home/zl/software/Genesis/examples/tutorials/mpm.py", line 70, in <module>
    scene.build()
  File "/home/zl/software/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/zl/software/Genesis/genesis/engine/scene.py", line 565, in build
    self._visualizer.build()
  File "/home/zl/software/Genesis/genesis/vis/visualizer.py", line 104, in build
    self._viewer.build(self._scene)
  File "/home/zl/software/Genesis/genesis/vis/viewer.py", line 69, in build
    self._pyrender_viewer = pyrender.Viewer(
  File "/home/zl/software/Genesis/genesis/ext/pyrender/viewer.py", line 378, in __init__
    self._initialized_event.wait()
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/threading.py", line 581, in wait
    signaled = self._cond.wait(timeout)
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
[Genesis] [12:11:39] [INFO] 💤 Exiting Genesis and caching compiled kernels...

(genesis) zl@zl:~/software/Genesis/examples/tutorials$ python hello_genesis.py 
[Genesis] [12:12:05] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [12:12:05] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [12:12:05] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [12:12:05] [INFO] Running on [13th Gen Intel(R) Core(TM) i9-13900HX] with backend gs.cpu. Device memory: 31.04 GB.
[Genesis] [12:12:05] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [12:12:05] [INFO] Scene <a53b59e> created.
[Genesis] [12:12:05] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <0504b4c>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [12:12:05] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <8d4f05d>, morph: <gs.morphs.MJCF(file='/home/zl/software/Genesis/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [12:12:05] [INFO] Building scene <a53b59e>...
[W 01/01/25 12:12:06.108 5405] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[... 21 lines cut ...]
    return XlibContext(self, share)
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context
^CTraceback (most recent call last):
  File "/home/zl/software/Genesis/examples/tutorials/hello_genesis.py", line 18, in <module>
    scene.build()
  File "/home/zl/software/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/zl/software/Genesis/genesis/engine/scene.py", line 565, in build
    self._visualizer.build()
  File "/home/zl/software/Genesis/genesis/vis/visualizer.py", line 104, in build
    self._viewer.build(self._scene)
  File "/home/zl/software/Genesis/genesis/vis/viewer.py", line 69, in build
    self._pyrender_viewer = pyrender.Viewer(
  File "/home/zl/software/Genesis/genesis/ext/pyrender/viewer.py", line 378, in __init__
    self._initialized_event.wait()
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/threading.py", line 581, in wait
    signaled = self._cond.wait(timeout)
  File "/home/zl/anaconda3/envs/genesis/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
[Genesis] [12:12:14] [INFO] 💤 Exiting Genesis and caching compiled kernels...

Many examples of sports report similar errors. How can I solve this problem? Thank you for anyone's help.

Hippozhibos, 2025-01-08:
Similar Problem:

`[Genesis] [16:19:58] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open swrast: /root/miniconda3/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/root/miniconda3/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/root/miniconda3/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/root/miniconda3/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1138, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/root/miniconda3/lib/python3.10/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
    super().__init__(*args, **kwargs)
  File "/root/miniconda3/lib/python3.10/site-packages/pyglet/window/__init__.py", line 527, in __init__
    context = config.create_context(gl.current_context)
  File "/root/miniconda3/lib/python3.10/site-packages/pyglet/gl/xlib.py", line 114, in create_context
    return XlibContext(self, share)
  File "/root/miniconda3/lib/python3.10/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context`

Hippozhibos, 2025-01-09:
Use "[Camera & Headless rendering](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html)" as substitute. I don't know the specific underlying reason for the bug, but it seems that using the Camera for rendering works better in headless environments.

tutar, 2025-02-23:
Similar problem：`libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri) libGL error: failed to load driver: swrast`

Env:
- WSL Ubuntu 22.04.5 LTS
- NVIDIA 3080
- python 3.12.9

solve:
```
%% 检查是否有swrast_dri.so %%
root: ls -al /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
-rw-r--r-- 13 root root 32612968 Jan 24  2024 /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so

%% 连接到/usr/lib/dri/swrast_dri.so %%
root: mkdir /usr/lib/dri
root: sudo ln -s /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so /usr/lib/dri/swrast_dri.so

%%替换libstdc++.so.6 %%
root: rm /opt/anaconda3/envs/genesis/lib/libstdc++.so.6
root: ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30 libstdc++.so.6
```

then i get the error:
` OpenGL.error.Error: Attempt to retrieve context when no valid context`
and solve [refer](https://github.com/Genesis-Embodied-AI/Genesis/issues/609#issuecomment-2603641579):
```
export MUJOCO_GL=glx
export PYOPENGL_PLATFORM=glx
```

closed 2025-02-28 by duburcqa