Genesis-Embodied-AI/genesis-world #432: Genesis Hello lottery 1 out of 5 - The lucky Ubuntu 24 machine

ai-bits, 2025-01-01:
I don't think that I need to throw a Xeon 256GB 2x 20GB RTX Ada at the Hello example, but that's the only that worked.

All machines >32GB, Anaconda Python env 2.12.8, PyTorch latest according to Nvidia (latest driver, CUDA 12.4) or not installed first, THEN Genesis from repo to account for numpy=1.26.4.

The visualization failed everywhere, except the Xeon with a repo ver from 3 days ago. Don't dare to pull from the repo or update the env.

On the M3 36GB MBP the visualization render_on_macos.py threw an error. Don't have time to dig it out now. (past midnight)

On the Razer 8GB RTX 3080 Ubuntu 24.04: pyglet.gl.ContextException: Could not create GL context.
Tried sudo apt install mesa-utils,.. Didn't help.

**On the Asus Duo Windows 11 latest w/o Nvidia it's the joke of the past year:**

**`[Genesis] [20:07:12] [ERROR] Camera not yet supported on Windows but is supposed to. Can anyone help?`**

I had tried to circumvent the non-existent viewer on Win by making a mp4, but the cam doesn't work either.

On the Duo Ubuntu 24.04 CPU-only **VMware** where I thought I could get a visualization: practically the same error as on the Razer. (see full below)

What a Happy New Year!
G.

genesis-error-duo-vm.txt

Genesis] [21:34:43] [INFO] Building visualizer...
MESA-LOADER: failed to open vmwgfx: /usr/lib/dri/vmwgfx_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: vmwgfx
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
Exception in thread Thread-3 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/gy/anaconda3/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/gy/anaconda3/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/gy/dl/Genesis/genesis/ext/pyrender/viewer.py", line 1138, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/gy/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
    super().__init__(*args, **kwargs)
  File "/home/gy/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/window/__init__.py", line 527, in __init__
    context = config.create_context(gl.current_context)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gy/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/gl/xlib.py", line 114, in create_context
    return XlibContext(self, share)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/gy/anaconda3/envs/genesis/lib/python3.12/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context

duburcqa, 2025-02-25:
Can you provide more details about the VMware setup that you were using ?

duburcqa, 2025-02-28:
I'm now considering this issue as fixed. I would be amazing if you could try again on all your devices with the main branch! Feel free to reopen if some issues are still present.

ai-bits, 2025-03-04:
Sorry it took me a while to get back, but since beginning of January when I filed several issues here a lot of other things happend and I needed to to finish them first.
And I had to reorient myself here.

The first thing I noticed:
[genesis-world.readthedocs.io](https://genesis-world.readthedocs.io/en/latest/index.html) > Quick Installation problem is still not fixed two months after:
As far as I remember, when you follow the instructions and `pip install genesis-world` and THEN install PyTorch some rendering lib (don't remember its name) gets overwritten and things don't work.

So please, before we go on to the other bugs, check in a new env if this is still the case or if I need to be more specific to repro this or it has resolved.

At the second glance, two months ago I reported that Prerequisites: Python: 3.9+ in [genesis-world.readthedocs.io](https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html) is a trap and should read
Python: >=3.10<3.13
because some items on the Mac need at least 3.10 and Genesis was incompatible with 3.13 - at least 2 months ago.

You could spare devs a lot of grieve and anger and time if you fixed this first.

Thanks for finally getting back.
G.

ai-bits, 2025-03-11:
Hey @duburcqa Alexis,
I'm waiting for you to verify and fix the two doc bugs before I go on testing the viewer alleged fixes on the different platforms.
Any time-frame?
Thanks
G.

duburcqa, 2025-03-22:
II will have a look to the documentation issue you mention and fix them if needed.

duburcqa, 2025-03-22:
After testing. Python version issue is real. No problem with PyTorch.

ai-bits, 2025-03-22:
OK, this is a good day.
I just checked my notes and luckily found that the _installation order must have been corrected already_. It says:
1) Install PyTorch following the [official instructions](https://pytorch.org/get-started/locally/).
2) Install Genesis via PyPI:

This is correct and should work.
Option: A note that the other way round does not work in this case (normally not an issue), but I don't want to be too picky.

Remains the Python version issue in [Prerequisites](https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html#prerequisites).
Please replace
**Python**: 3.9+
with
**Python**: >=3.9<3.13 (Mac: >=3.10<3.13)
or similar and close the issue again.

I will then go on with checking the state of the laptop Optimus integrated graphics v Nvidia issue and the Mac viewer.
Thanks and have a good night!
G.

duburcqa, 2025-03-22:
> A note that the other way round does not work in this case (normally not an issue)

I disagree. It works whatever the order.

> Remains the Python version issue

Yes, update in progress, but it will take some time to propagate on the official doc website.

ai-bits, 2025-03-22:
>> A note that the other way round does not work in this case (normally not an issue)

>I disagree. It works whatever the order on the fresh environment I tested this morning. At least for main branch.

OK, if you verified. The times they are a changin'. Guess when I checked it was around Xmas. ;-)

Cheers
G.

closed 2025-03-23 by ai-bits