Genesis-Embodied-AI/genesis-world #435: online rigid  operating

Xu-Xuanhui, 2025-01-02:
Can I combine two rigid bodies into one when certain conditions are met during simulation, and then separate them when they are not met?

zhouxian, 2025-01-02:
Not yet. This is a popular request and is needed for applications like suction cup. We are working on it

Xu-Xuanhui, 2025-01-02:
> Not yet. This is a popular request and is needed for applications like suction cup. We are working on it

Thank you for your valuable work and I am looking forward to the future versions of Genesis.

Xu-Xuanhui, 2025-01-02:
> Not yet. This is a popular request and is needed for applications like suction cup. We are working on it
May I ask in which version this feature will be updated？

Xu-Xuanhui, 2025-01-02:
> Not yet. This is a popular request and is needed for applications like suction cup. We are working on it

May I ask in which version this feature will be updated

zhouxian, 2025-01-02:
Very soon. I’d say 0.2.3?

YilingQiao, 2025-03-31:
This feature is added #961