Genesis-Embodied-AI/genesis-world #442: visualizer lottery 2 out of 5 - educated owner of Optimus laptop on Ubuntu forces Nvidia graphics

ai-bits, 2025-01-02:
After a good night's sleep it occurred to me that the `pyglet.gl.ContextException: Could not create GL context` on the Razer RTX 3080 laptop was due to (Optimus) staying on the Intel integrated GPU.

Forcing the RTX 3080 in Nvidia X Server Settings > PRIME Profiles from Nvidia On-Demand to Nvidia Performance Mode did the trick.

So, if Genesis does not auto-switch to the discrete performance GPU like any other graphics-heavy app, it should handle it more gracefully / with a less cryptic exception.

Cheers
G.

zhouxian, 2025-01-02:
Is this on windows?

ai-bits, 2025-01-02:
Nope, as the title says, Ubuntu (24.04) and there clearly are no  Nvidia **X Server Settings** on Windows (or WSL) and I do read the docs that visualization is not yet supported on Windoze, but I should have said it in the text too.

Thanks
G.

zhouxian, 2025-01-02:
I see. I am on my phone so I didn’t read it carefully

We didn’t test thoroughly on 24.04

backend is for simulation only. Rendering will use gl regardless

we will test on 24.04 and get back

ai-bits, 2025-01-02:
I don't think this has _anything_ to do with Ubuntu versions, though on Ubuntu they switched between X11 and Wayland LIKE CRAZY.

I'm afraid ANY **(laptop)** user on ANY Linux with a CPU-integrated Intel GPU AND an Nvidia GPU and Optimus (auto-)switching will run into this **and that's a LOT of users!**

duburcqa, 2025-02-25:
I can have a look but I don't have a setup to reproduce the issue locally. Would you be able to check if a PR fixes your issue ?

closed 2025-03-22 by duburcqa