Genesis-Embodied-AI/genesis-world #445: Illegal instruction (core dumped)

git523, 2025-01-02:
Has anyone came across this error message before? I was able to successfully install the program, but when I ran the examples, this single line of error message popped up: "Illegal instruction (core dumped)".

Kashu7100, 2025-01-03:
Can you provide:
1. your system setup
2. full program you run
3. full error msg

git523, 2025-01-03:
Thank you very much for help looking into this. I used Ubuntu 24.04.1 in virtual box, and the settings in virtualbox are shown in the screenshots below, along with the error message that i got when I ran the sample workflow. I also attached the virtualbox VBoxHardening.log, just in case it is useful...

[image]
[image]
[image]

VBoxHardening.log

=================
32e4.11f8: \SystemRoot\System32\ntdll.dll:
32e4.11f8:     CreationTime:    2024-12-13T07:29:03.608119400Z
32e4.11f8:     LastWriteTime:   2024-12-13T07:29:03.637118400Z
32e4.11f8:     ChangeTime:      2024-12-13T09:01:00.981543700Z
32e4.11f8:     FileAttributes:  0x20
32e4.11f8:     Size:            0x216050
32e4.11f8:     NT Headers:      0xe8
32e4.11f8:     Timestamp:       0xe7035eba
32e4.11f8:     Machine:         0x8664 - amd64
32e4.11f8:     Timestamp:       0xe7035eba
32e4.11f8:     Image Version:   10.0
32e4.11f8:     SizeOfImage:     0x217000 (2191360)
32e4.11f8:     Resource Dir:    0x1a0000 LB 0x759a8
32e4.11f8:     [Version info resource found at 0xd8! (ID/Name: 0x1; SubID/SubName: 0x409)]
32e4.11f8:     [Raw version resource data: 0x1a00f0 LB 0x380, codepage 0x0 (reserved 0x0)]
32e4.11f8:     ProductName:     Microsoft® Windows® Operating System
32e4.11f8:     ProductVersion:  10.0.22621.4541
32e4.11f8:     FileVersion:     10.0.22621.4541 (WinBuild.160101.0800)
32e4.11f8:     FileDescription: NT Layer DLL
32e4.11f8: \SystemRoot\System32\kernel32.dll:
32e4.11f8:     CreationTime:    2024-11-13T05:59:53.499390900Z
32e4.11f8:     LastWriteTime:   2024-11-13T05:59:53.515391500Z
32e4.11f8:     ChangeTime:      2024-12-13T07:29:48.132715800Z
32e4.11f8:     FileAttributes:  0x20
32e4.11f8:     Size:            0xc71e0
32e4.11f8:     NT Headers:      0xe8
32e4.11f8:     Timestamp:       0x1ef15383
32e4.11f8:     Machine:         0x8664 - amd64
32e4.11f8:     Timestamp:       0x1ef15383
32e4.11f8:     Image Version:   10.0
32e4.11f8:     SizeOfImage:     0xc4000 (802816)
32e4.11f8:     Resource Dir:    0xc2000 LB 0x520
32e4.11f8:     [Version info resource found at 0x90! (ID/Name: 0x1; SubID/SubName: 0x409)]
32e4.11f8:     [Raw version resource data: 0xc20b0 LB 0x3a4, codepage 0x0 (reserved 0x0)]
32e4.11f8:     ProductName:     Microsoft® Windows® Operating System
32e4.11f8:     ProductVersion:  10.0.22621.4391
32e4.11f8:     FileVersion:     10.0.22621.4391 (WinBuild.160101.0800)
32e4.11f8:     FileDescription: Windows NT BASE API Client DLL
32e4.11f8: \SystemRoot\System32\KernelBase.dll:
32e4.11f8:     CreationTime:    2024-12-13T07:29:04.025349500Z
32e4.11f8:     LastWriteTime:   2024-12-13T07:29:04.117348800Z
32e4.11f8:     ChangeTime:      2024-12-13T09:01:00.557541200Z
32e4.11f8:     FileAttributes:  0x20
32e4.11f8:     Size:            0x3c0bb0
32e4.11f8:     NT Headers:      0xf8
32e4.11f8:     Timestamp:       0x8ca6fab8
32e4.11f8:     Machine:         0x8664 - amd64
32e4.11f8:     Timestamp:       0x8ca6fab8
32e4.11f8:     Image Version:   10.0
32e4.11f8:     SizeOfImage:     0x3ba000 (3907584)
32e4.11f8:     Resource Dir:    0x389000 LB 0x548
32e4.11f8:     [Version info resource found at 0x90! (ID/Name: 0x1; SubID/SubName: 0x409)]
32e4.11f8:     [Raw version resource data: 0x3890b0 LB 0x3bc, codepage 0x0 (reserved 0x0)]
32e4.11f8:     ProductName:     Microsoft® Windows® Operating System
32e4.11f8:     ProductVersion:  10.0.22621.4541
32e4.11f8:     FileVersion:     10.0.22621.4541 (WinBuild.160101.0800)
32e4.11f8:     FileDescription: Windows NT BASE API Client DLL
32e4.11f8: \SystemRoot\System32\apisetschema.dll:
32e4.11f8:     CreationTime:    2024-08-18T04:48:46.413130700Z
32e4.11f8:     LastWriteTime:   2024-08-18T04:48:46.415247000Z
32e4.11f8:     ChangeTime:      2024-12-13T07:29:48.046715800Z
[... 437 lines cut ...]
32e4.11f8:   00007ffefa4a1e43 / 0x0271e43: fa != fc
32e4.11f8:   Restored 0x2000 bytes of original file content at 00007ffefa4a1000
32e4.11f8: supHardNtVpCheckHandles:
32e4.11f8: supR3HardenedWinInit: SUPHARDNTVPKIND_SELF_PURIFICATION_LIMITED -> VINF_SUCCESS, cFixes=5
32e4.11f8: '\Device\HarddiskVolume3\Program Files\Oracle\VirtualBox\VirtualBoxVM.exe' has no imports
32e4.11f8: supHardenedWinVerifyImageByHandle: -> 0 (\Device\HarddiskVolume3\Program Files\Oracle\VirtualBox\VirtualBoxVM.exe)
32e4.11f8: supR3HardNtEnableThreadCreationEx:
32e4.11f8: supR3HardNtDisableThreadCreation: pvLdrInitThunk=00007ffefcb84400 pvNtTerminateThread=00007ffefcbb0df0
32e4.11f8: supR3HardenedWinDoReSpawn(1): New child 1980.3028 [kernel32].
32e4.11f8: supR3HardNtChildGatherData: PebBaseAddress=000000dfb8814000 cbPeb=0x388
32e4.11f8: supR3HardNtPuChFindNtdll: uNtDllParentAddr=00007ffefcb10000 uNtDllChildAddr=00007ffefcb10000
32e4.11f8: supR3HardenedWinSetupChildInit: uLdrInitThunk=00007ffefcb84400
32e4.11f8: supR3HardenedWinSetupChildInit: Initial context:
  rax=0000000000000000 rbx=0000000000000000 rcx=00007ff7e865b790 rdx=000000dfb8814000
  rsi=0000000000000000 rdi=0000000000000000 r8 =0000000000000000 r9 =0000000000000000
  r10=0000000000000000 r11=0000000000000000 r12=0000000000000000 r13=0000000000000000
  r14=0000000000000000 r15=0000000000000000  P1=0000000000000000  P2=0000000000000000
  rip=00007ffefcb6af10 rsp=000000dfb8aff9f8 rbp=0000000000000000    ctxflags=0010001b
  cs=0033 ss=002b ds=0000 es=0000 fs=0000 gs=0000    eflags=00000200   mxcrx=00001f80
   P3=0000000000000000  P4=0000000000000000  P5=0000000000000000  P6=0000000000000000
  dr0=0000000000000000 dr1=0000000000000000 dr2=0000000000000000 dr3=0000000000000000
  dr6=0000000000000000 dr7=0000000000000000 vcr=0000000000000000 dcr=0000000000000000
  lbt=0000000000000000 lbf=0000000000000000 lxt=0000000000000000 lxf=0000000000000000
32e4.11f8: supR3HardenedWinSetupChildInit: Start child.
32e4.11f8: supR3HardNtChildWaitFor: Found expected request 0 (PurifyChildAndCloseHandles) after 0 ms.

git523, 2025-01-05:
A quick follow-up on the error message: I installed Anaconda in Ubuntu, and then installed Genesis within Anaconda, re-ran the pbd_liquid.py example code, and still not successful. But this time I was able to get the error message. Please see below the screenshots of the error message (sorry I was not able to copy out the text somehow):
[image]
[image]

git523, 2025-01-10:
Hey @Kashu7100, could you help take a look at the error messages when you have chance? Thank you!

Kashu7100, 2025-01-10:
Sorry for the late reply. I am not familiar with the VMs (also with spyder) actually.
Is the kernel thing related to spyder? If so, can you just use the normal python to see what happens?

git523, 2025-01-10:
Sure no worries @Kashu7100, thanks and I will try the alternative approach.

CarlosMagni, 2025-01-13:
Same issue on a VM with proxmox as hypervisor

closed 2025-02-20 by Kashu7100