Genesis-Embodied-AI/genesis-world #446: No display detected. Use `show_viewer=False` for headless mode

oupuxi, 2025-01-02:
SORRY TO BOTHER GUYS! I'm encountering an error when running my first program, hello_genesis.py. I am using a rented server without a monitor. Below is the error log I received.
/root/.virtualenvs/Genesis-main/bin/python /root/autodl-tmp/genesis/Genesis-main/examples/tutorials/hello_genesis.py 
[Genesis] [20:41:45] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [20:41:45] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [20:41:45] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [20:41:45] [INFO] Running on [Intel(R) Xeon(R) Platinum 8481C] with backend gs.cpu. Device memory: 1007.54 GB.
[Genesis] [20:41:45] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/root/autodl-tmp/genesis/Genesis-main/examples/tutorials/hello_genesis.py", line 5, in <module>
    scene = gs.Scene()
            ^^^^^^^^^^
  File "/root/autodl-tmp/genesis/Genesis-main/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/root/autodl-tmp/genesis/Genesis-main/genesis/engine/scene.py", line 147, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/root/autodl-tmp/genesis/Genesis-main/genesis/vis/visualizer.py", line 37, in __init__
    gs.raise_exception("No display detected. Use `show_viewer=False` for headless mode.")

[Genesis] [20:41:46] [ERROR] No display detected. Use `show_viewer=False` for headless mode.
[Genesis] [20:41:46] [INFO] 💤 Exiting Genesis and caching compiled kernels...

Process finished with exit code 1

zhouxian, 2025-01-02:
So you will need to set show viewer to False as the error msg says, since you are not connected to a monitor

oupuxi, 2025-01-03:
> 因此，您需要将 show viewer 设置为 False，如错误 msg 所示，因为您没有连接到监视器

I know this is possible, since it's an experiment running on a cloud server, but I wonder if there's any way to display it on my own computer.

oupuxi, 2025-01-03:
> So you will need to set show viewer to False as the error msg says, since you are not connected to a monitor

I know this is possible, since it's an experiment running on a cloud server, but I wonder if there's any way to display it on my own computer.

mryangs, 2025-01-03:
> > So you will need to set show viewer to False as the error msg says, since you are not connected to a monitor因此，您需要将show viewer设置为False，因为您没有连接到监视器，
> 
> I know this is possible, since it's an experiment running on a cloud server, but I wonder if there's any way to display it on my own computer.我知道这是可能的，因为它是在云服务器上运行的实验，但我想知道是否有任何方法可以在我自己的计算机上显示它。

I have also encountered the same problem. Have you solved it yet?

duburcqa, 2025-03-22:
All known display issues have been fixed on main branch. Feel free to reopen if still not working.

closed 2025-03-22 by duburcqa