Genesis-Embodied-AI/genesis-world #447: what does quat mean in genesis

milong26, 2025-01-02:
I searched that quaternion wxyz means rotation. so (1,0,0,0) means keep same direction while (0,1,0,0) means 180degree rotation about y-axis.

Then why in [ik and mp tuorial](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/inverse_kinematics_motion_planning.html) you set (0,1,0,0) when # reach

I can't understand these code ;-; so sorry for bothering u

Code is here:

```python
# move to pre-grasp pose
qpos = franka.inverse_kinematics(
    link = end_effector,
...
    quat = np.array([0, 1, 0, 0]),
)
...
# reach
qpos = franka.inverse_kinematics(
    link = end_effector,
...
    quat = np.array([0, 1, 0, 0]),
)

...
# why keep same quat?when i try to use(1,0,0,0) it rotated 180 degree about y-axis ;-;
```

oscar-lima, 2025-01-02:
Likely they’re using a convention where the quaternion is in the order ```[x, y, z, w]``` which is common in robotics. In that format, ```[0, 1, 0, 0]``` happens to be the "neutral" orientation for the Franka’s end effector in this scene. Swapping it to ```[1, 0, 0, 0]``` flips the gripper 180° about the y-axis. That’s why the code consistently uses ```[0, 1, 0, 0]``` for all poses. It is a bit strange that neither the code or the documentation clearly mentions which quaternion convention they expect for which function... because if you grep -r for xyzw and wxyz with their repo you get plenty of results for both, suggesting that for some methods use xyzw and others wxyz, they even have converter functions, see e.g., ```genesis/utils/geom.py:def wxyz_to_xyzw(wxyz):```. Upon further review they seem to store quaternions internally as [w, x, y, z], but in many places (such as inverse_kinematics) you provide ```[x, y, z, w]``` and the code converts it behind the scenes.

milong26, 2025-01-03:
> Likely they’re using a convention where the quaternion is in the order `[x, y, z, w]` which is common in robotics for that particular function. In that format, `[0, 1, 0, 0]` happens to be the "neutral" orientation for the Franka’s end effector in this scene. Swapping it to `[1, 0, 0, 0]` flips the gripper 180° about the y-axis. That’s why the code consistently uses `[0, 1, 0, 0]` for all poses. If true, is a bit strange that neither the code or the documentation clearly mentions which quaternion convention they expect for that function... if you grep -r for xyzw and wxyz with their repo you get plenty of results for both, suggesting that for some methods use xyzw and others wxyz, they even have converter functions, see e.g., `genesis/utils/geom.py:def wxyz_to_xyzw(wxyz):`. Upon further review they seem to store quaternions internally as `[w, x, y, z]`, but in many places (such as inverse_kinematics) apparently you provide `[x, y, z, w]` and the code converts it behind the scenes.

xD ye i remember I saw wxyz in their document before..
So we can only stay here and wait for their explanation

zhouxian, 2025-01-03:
Yes we use wxyz

we used (0,1,0,0) in the example because you would want the gripper to point down. if it's (1,0,0,0) it points upward

closed 2025-01-03 by zhouxian

milong26, 2025-01-03:
> Yes we use wxyz
> 
> we used (0,1,0,0) in the example because you would want the gripper to point down. if it's (1,0,0,0) it points upward

Does (w,x,y,z) means w rotation(w=cos(theta/2)) around (x,y,z) axis? Then why in # reach part it do (0,1,0,0) again,since it stands for rotate 180degree around (1,0,0) axis=x axis