Genesis-Embodied-AI/genesis-world #448: error in docker build stage

serg-yalosovetsky, 2025-01-02:
echo "# Issue: \`build_luisa.sh\` Not Found During Docker Build

## Description
While building the Docker image, the script \`build_luisa.sh\` fails with the error:

\`\`\`
/bin/sh: 1: ./build_luisa.sh: not found
\`\`\`

## Steps to Reproduce
1. Run the following Docker build command:

   \`\`\`bash
   docker build -t genesis -f docker/Dockerfile docker
   \`\`\`

2. The error occurs during the execution of the following lines in the Dockerfile:

   \`\`\`dockerfile
   WORKDIR /workspace
   RUN git clone https://github.com/Genesis-Embodied-AI/Genesis.git && \\
       cd Genesis && \\
       git submodule update --init --recursive
   COPY build_luisa.sh /workspace/build_luisa.sh
   RUN chmod +x ./build_luisa.sh && ./build_luisa.sh \${PYTHON_VERSION}
   \`\`\`

## Expected Behavior
The script \`build_luisa.sh\` should be executed without any errors.

## Observed Behavior
The Docker build fails with the error \`/bin/sh: 1: ./build_luisa.sh: not found\`.

## Debugging Steps Taken
To debug the issue, the following steps were added to the Dockerfile:

\`\`\`dockerfile
RUN pwd && ls -l /workspace && ls -l /workspace/build_luisa.sh && \\
    echo \"File location: \$(realpath ./build_luisa.sh)\" && \\
    chmod +x ./build_luisa.sh && ./build_luisa.sh \${PYTHON_VERSION}
\`\`\`

These debug commands show that:

- The script is present in \`/workspace/\`.
- The full path of the script is printed as expected.

## Additional Context
- The \`build_luisa.sh\` file is copied from the local directory (\`docker/build_luisa.sh\`).
- The issue may be related to the context in which the script is being executed or the working directory after cloning the repository.

[error.md](https://github.com/user-attachments/files/18291054/error.md)

Kashu7100, 2025-01-03:
If the script is present in `/workspace`, could you check if you can run:
```
RUN chmod +x /workspace/build_luisa.sh && /workspace/build_luisa.sh ${PYTHON_VERSION}
```

liscinskymartin, 2025-01-03:
You can fix it by changing CRLF line editing to LF line editing in VSCode. Please, check #397