Genesis-Embodied-AI/genesis-world #456: Bug. ValueError: Incorrect texture coordinate shape

Riften, 2025-01-03:
**Description**
Error occurs when build scene with MJCF loaded. `ValueError: Incorrect texture coordinate shape`

<details>
<summary>Full error message</summary>

```bash
[Genesis] [14:03:01] [INFO] ╭───────────────────────────────────────────────────────────────────────────────╮
[Genesis] [14:03:01] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [14:03:01] [INFO] ╰───────────────────────────────────────────────────────────────────────────────╯
[Genesis] [14:03:01] [INFO] Running on [NVIDIA GeForce GTX 1080 Ti] with backend gs.cuda. Device memory: 10.91 GB.
[Genesis] [14:03:01] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [14:03:02] [INFO] Scene <f2f61d3> created.
[Genesis] [14:03:02] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <2234a0c>, morph: <gs.morphs.MJCF(file='/home/yongxi/Workspace/SJTU-Robotic-Interface/notebooks/cup_2/model.xml')>, material: <gs.materials.Rigid>.
[Genesis] [14:03:02] [INFO] Building scene <f2f61d3>...
[Genesis] [14:03:12] [INFO] Compiling simulation kernels...
[Genesis] [14:03:45] [INFO] Building visualizer...
Traceback (most recent call last):
  File "/home/yongxi/Workspace/SJTU-Robotic-Interface/notebooks/bug_value_error.py", line 6, in <module>
    scene.build()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/scene.py", line 607, in build
    self._visualizer.build()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/vis/visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/vis/rasterizer_context.py", line 91, in build
    self.on_rigid()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/vis/rasterizer_context.py", line 263, in on_rigid
    pyrender.Mesh.from_trimesh(
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/mesh.py", line 222, in from_trimesh
    Primitive(
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/primitive.py", line 85, in __init__
    self.texcoord_0 = texcoord_0
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/primitive.py", line 158, in texcoord_0
    raise ValueError("Incorrect texture coordinate shape")
ValueError: Incorrect texture coordinate shape
[Genesis] [14:03:48] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

</details>

**To Reproduce**
code

```python
import genesis as gs
gs.init(backend=gs.cuda)
scene = gs.Scene()
test_mjcf_file = 'cup_2/model.xml'
scene.add_entity(gs.morphs.MJCF(file=test_mjcf_file))
scene.build()
```

resource

[cup_2.zip](https://github.com/user-attachments/files/18296890/cup_2.zip)

**Expected behavior**

Load the same mjcf model to mujoco 3.2.5

[image]

**Environment:**
 - OS: Ubuntu 20.04
 - GPU/CPU: NVIDIA GeForce GTX 1080 Ti

zhouxian, 2025-01-03:
ok we will take a look

Riften, 2025-01-03:
Here is another ValueError when load a different MJCF model. `ValueError: Cannot reformat 2-channel texture into RGBA`.

Here is the model file.

[outlet_room.zip](https://github.com/user-attachments/files/18298215/outlet_room.zip)

Here is the result in mujoco 3.2.5

[image]

<details>

<summary>Full error message</summary>

```bash
[Genesis] [17:30:48] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [17:30:48] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [17:30:48] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [17:30:48] [INFO] Running on [NVIDIA GeForce GTX 1080 Ti] with backend gs.cuda. Device memory: 10.91 GB.
[Genesis] [17:30:48] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [17:30:49] [INFO] Scene <4d5a5c8> created.
[Genesis] [17:30:49] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <7405e0b>, morph: <gs.morphs.MJCF(file='/home/yongxi/Workspace/SJTU-Robotic-Interface/notebooks/outlet_room/model.xml')>, material: <gs.materials.Rigid>.
[Genesis] [17:30:49] [INFO] Building scene <4d5a5c8>...
[Genesis] [17:30:56] [INFO] Compiling simulation kernels...
[Genesis] [17:31:43] [INFO] Building visualizer...
Traceback (most recent call last):
  File "/home/yongxi/Workspace/SJTU-Robotic-Interface/notebooks/bug_value_error_2.py", line 6, in <module>
    scene.build()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/scene.py", line 607, in build
    self._visualizer.build()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/vis/visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/vis/rasterizer_context.py", line 91, in build
    self.on_rigid()
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/vis/rasterizer_context.py", line 263, in on_rigid
    pyrender.Mesh.from_trimesh(
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/mesh.py", line 205, in from_trimesh
    color_0, texcoord_0, primitive_material = Mesh._get_trimesh_props(m, smooth=smooth, material=material)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/mesh.py", line 310, in _get_trimesh_props
    material = MetallicRoughnessMaterial(
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/material.py", line 398, in __init__
    self.baseColorTexture = baseColorTexture
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/material.py", line 421, in baseColorTexture
    self._baseColorTexture = self._format_texture(value, "RGBA")
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/material.py", line 268, in _format_texture
    source = format_texture_source(texture, target_channels)
  File "/home/yongxi/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/ext/pyrender/utils.py", line 98, in format_texture_source
    raise ValueError("Cannot reformat 2-channel texture into RGBA")
ValueError: Cannot reformat 2-channel texture into RGBA
[Genesis] [17:31:46] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

</details>

koukoucoconut, 2025-01-07:
same situation. Do you solve it?

Riften, 2025-01-09:
@zhenjia-xu It works. Thank you!

closed 2025-01-09 by Riften