Genesis-Embodied-AI/genesis-world #461: mjcf parser wrong

DTSwink, 2025-01-03:
im trying to repurpose the dog walk to a mjcf humanoid example and noticed some weird stuff
firstly , in rigid_entity.py  get_joint() there is a "`joint.id`" reference that is undefined, by checking the base class i think it should be "`idx`", that works
But the bigger issue is that it doesnt seem to loop on all joints, and get some weird repetition
here is the log i get for ```
for joint in self._joints:
                print("joint_name =" ,joint.name,"| joint_idx=",joint.idx)
```

`joint_name = root  | joint_idx= 1
joint_name = abdomen_x  | joint_idx= 2
joint_name = root  | joint_idx= 1
joint_name = abdomen_x  | joint_idx= 2
joint_name = right_hip_x  | joint_idx= 3
joint_name = left_hip_x  | joint_idx= 4
joint_name = neck_x  | joint_idx= 5
joint_name = right_shoulder_x  | joint_idx= 6
joint_name = left_shoulder_x  | joint_idx= 7
joint_name = right_knee  | joint_idx= 8
joint_name = left_knee  | joint_idx= 9
joint_name = right_elbow  | joint_idx= 10
joint_name = left_elbow  | joint_idx= 11
joint_name = right_ankle_x  | joint_idx= 12
joint_name = left_ankle_x  | joint_idx= 13
joint_name = right_hand_joint  | joint_idx= 14
joint_name = left_hand_joint  | joint_idx= 15`

ive attached the .xml just in case 

[human.txt](https://github.com/user-attachments/files/18298321/human.txt)

DTSwink, 2025-01-03:
i do get these warnings, i dont know if that's related :

`[Genesis] [10:50:08] [WARNING] Overriding base link's pos with user provided value in morph.
[Genesis] [10:50:08] [WARNING] Overriding base link's quat with user provided value in morph.
[Genesis] [10:50:09] [WARNING] Overriding base link's pos with user provided value in morph.
[Genesis] [10:50:09] [WARNING] Overriding base link's quat with user provided value in morph.`

DTSwink, 2025-01-03:
so i tried on the repo's `ant.xml` for reference
the `id` -> `idx` point still stands, however the joints are all listed correctly, so something is not right either with my .xml or the way it is parsed
i'll report if i find something useful

zhouxian, 2025-01-03:
id should actually be uid. We refactored the name at some point but probably missed some 

I am not sure if this is relevant though. We will take a look

DTSwink, 2025-01-03:
well the get_joint() function accept a parameter that doesnt exist and return an error so idk
the parameter in itself doesnt look crucial at first glance i would agree
im still wondering wondering why my xml can't load, i'll report on my findings i know you must be quite busy lately

DTSwink, 2025-01-03:
i tried with the DeepMind humanoid.xml and same problem, some joints are listed by get_joint(), some are ommitted
that's more of a serious issue i would argue.. will try random tweaks until works

zhouxian, 2025-01-03:
Sounds like a bug to me

DTSwink, 2025-01-04:
yes so indeed, genesis when parsing mjcf .xml files assumes there isn't multiple joints per body, i believe in `rigid_entity.py` in `_load_MJCF` :

```
 for i_l in range(n_links):
            #el problemo ->
            l_info, j_info = mju.parse_link(mj, i_l + 1, q_offset, dof_offset, morph.scale)
            
            l_infos.append(l_info)
            j_infos.append(j_info)
            

            q_offset += j_info["n_qs"]
            dof_offset += j_info["n_dofs"]
```

where `n_links = mj.nbody - 1 `
if number of joints exceed number of body -1 then some will be omitted

i wish i could solve this but it involves several deep function in genesis and it goes currently above my head, hopefully the team can fix it
in the mean time i'll try the URDF road see if i get more luck
cheers everyone

DTSwink, 2025-01-05:
Update : the training runs but the simulation appears empty, similarly to https://github.com/Genesis-Embodied-AI/Genesis/issues/287 unfortunately

[image]

DTSwink, 2025-01-05:
alright here are my findings so far , putting this at the start of the .xml make the mesh appear
```
<default>
        <joint armature="1" damping="1" limited="true"/>
        <geom contype="0" conaffinity="0" condim="3" density="5.0" friction="1 0.5 0.5" rgba="0.8 0.6 0.4 1"/>
</default>
```

i believe it is the `contype="0" conaffinity="0"` that does the trick
now the mesh dont have collision, if u set 'contype=1' on a geom it disappear, for a mesh to show and have collision you need to duplicate the geom and set `contype=1` to the duplicate like so :

```
 <geom name="sphere1_geom" size="0.08" type="sphere" pos="0 0 1.0" />
 <geom name="sphere1_geom_c" contype="1" size="0.08" type="sphere" pos="0 0 1.0" />
```
dont ask me why, maybe it's standard mujoco practice, maybe im doing it wrong, im just a noob tweaking things and reporting my findings, til next report 
regards 
[image]

DTSwink, 2025-01-05:
note : as pointed in a other issue `box` geom are not supported yet so it gotta be transformed into capsule or spheres

kolergy, 2025-01-05:
Thanks @DTSwink had similar issue and adding:
`<default>
        <joint armature="1" damping="1" limited="true"/>
        <geom contype="0" conaffinity="0" condim="3" density="5.0" friction="1 0.5 0.5" rgba="0.8 0.6 0.4 1"/>
</default>`
Worked perfectly

DTSwink, 2025-01-05:
yooo yall are reading this :O 
@kolergy glad it helped, cheers ;)

milong26, 2025-01-06:
> i do get these warnings, i dont know if that's related :
>
> ```
>
> [Genesis] [10:50:08] [WARNING] Overriding base link's pos with user provided value in morph.
>
> [Genesis] [10:50:08] [WARNING] Overriding [...]

THis is just a warning when you want to use user-defined quat or pos

DTSwink, 2025-01-06:
> THis is just a warning when you want to use user-defined quat or pos

yeah i messed up i meant to quote these warnings :
`[Genesis] [19:59:34] [WARNING] kp in gainprm doesn't match -kp in biasprm.`

for anyone having these and seeing no joint activity it might be related, i fixed it by changing "motor" actuators by "position" actuators and setting `kp=1` like so :

```
<actuator>
    <position ctrllimited="true" ctrlrange="-1.0 1.0" joint="hip_4" gear="150" kp="1" kv="0.1"/>
</actuator>
```

duburcqa, 2025-03-22:
This issue has been fixed on main branch.

closed 2025-03-22 by duburcqa