Genesis-Embodied-AI/genesis-world #462: Error:   RRTConnect: Motion planning start tree could not be initialized!          at line 219 in /project/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp

milong26, 2025-01-03:
I encountered this error when i run motion plan like this
```python
path1 = franka1.plan_path(
    qpos_goal     = qpos1,
    num_waypoints = 200, 
)
for waypoint in path1:
    franka1.control_dofs_position(waypoint)
    scene.step()
    cam.render()
# franka1.control_dofs_position(qpos1[:-2], motors_dof1)
```
the qpos are not problem becuase when I run `franka1.control_dofs_position(qpos1[:-2], motors_dof1)`  it works (but a bit strange) so can u tell me when this error raised?

The whole log is
```
Error:   RRTConnect: Motion planning start tree could not be initialized!
         at line 219 in /project/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp
[Genesis] [18:50:13] [WARNING] Path planning failed. Returning empty path.
```
Thanks!

zswang666, 2025-01-06:
This probably means that you have an invalid start state. Suggested debugging steps:
- Turn on either `ignore_joint_limit` or `ignore_collision` and check if any of the combination works.
- If at least one of the case in the previous step works, identify the cause of invalid initial state -- your robot either (i) violates joint limit or (ii) have contact with other entities in the scene.
- Revise your initial state accordingly.

DengyingFu, 2025-02-24:
> This probably means that you have an invalid start state. Suggested debugging steps:
>
>     * Turn on either `ignore_joint_limit` or `ignore_collision` and check if any of the combination works. [...]

Does this mean that when performing the task of grasping the robotic arm？the path cannot be planned after grabbing the object (my current situation seems to be like this, after grabbing something, an error is reported, and the ignore_collision parameter is valid)

milong26, 2025-02-25:
> Does this mean that when performing the task of grasping the robotic arm？the path cannot be planned after grabbing the object (my current situation seems to be like this, after grabbing something, [...]

You should set ignore_collision because the arm will think the object it grabbed is collision too

closed 2025-02-25 by milong26