Genesis-Embodied-AI/genesis-world #465: LuisaRenderPy won't compile with NVTT enabled

kaspar-gestalt, 2025-01-03:
In order to render a scene with spherical light sources, I am using this code:

```
import genesis as gs
gs.init(backend=gs.gpu)

lights = [{'type': 'directional', 'dir': (-1, -1, -1), 'color': (1.0, 1.0, 1.0), 'intensity': 5.0},
       {'type': 'directional', 'dir': (2,2, 1), 'color': (0.5, 1.0, 1.0), 'intensity': 2.0}, ]

scene = gs.Scene(show_viewer=False,
        vis_options = gs.options.VisOptions(
            show_world_frame = False,
            segmentation_level='entity',
            world_frame_size = 1.0,
            show_link_frame  = False,
            show_cameras     = False,
            plane_reflection = True,
            ambient_light    = (0.1, 0.1, 0.1),
            lights=lights),
        renderer=gs.renderers.RayTracer(),
)
motor = scene.add_entity(gs.morphs.Mesh(file='/path/to/my/object.obj'))
franka = scene.add_entity(gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml', pos=(1,0,0)))
box = scene.add_entity(gs.morphs.Box(pos=(0,0.5,0), size=(0.1,0.2,0.1)))

cam = scene.add_camera(
    res    = (1280, 960),
    pos    = (2,0,0),
    lookat = (0, 0, 0),
    fov    = 30,
    GUI    = False
)
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                            
scene.add_light(gs.morphs.Sphere(pos=(0,-1.4,-1.4), radius=0.1), color=(0,0,1,1), intensity=20)                                                                                                                                                                                                                                                                                                                               
scene.build()                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                    
                                                                                                                                                                                                                                                                                                                                                                                                                                    
rgb, depth, segmentation, normal = cam.render(depth=True,colorize_seg=False, segmentation=True, normal=True)                                                                                                                                                                                                                                                                                                                    
```

I compiled LuisaRenderPy following the [official instructions](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#setup), including the CUDA backend. I then got an error:

```
2025-01-02 10:37:18.581] [console] [error] CUDA_ERROR_INVALID_SOURCE: device kernel image is invalid [/home/kaspar/repos/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/cuda_shader_native.cpp:46]
     0 [0x7fd9eb55589d]: /home/kaspar/repos/Genesis/genesis/ext/LuisaRender/build/bin/liblc-backend-cuda.so :: unknown + 1325213
     1 [0x7fd9eb5563ce]: /home/kaspar/repos/Genesis/genesis/ext/LuisaRender/build/bin/liblc-backend-cuda.so :: luisa::compute::cuda::CUDAShaderNative::CUDAShaderNative(luisa::compute::cuda::CUDADevice*, eastl::vector<std::byte, eastl::allocator>, char const*, luisa::compute::cuda::CUDAShaderMetadata const&, eastl::vector<luisa::compute::Argument, eastl::allocator>) + 510
     2 [0x7fd9eb4c3c9a]: /home/kaspar/repos/Genesis/genesis/ext/LuisaRender/build/bin/liblc-backend-cuda.so :: luisa::compute::cuda::CUDADevice::_create_shader(std::__cxx11::basic_string<char, std::char_traits<char>, luisa::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, luisa::allocator<char> > const&, luisa::compute::ShaderOption const&, eastl::span<char const* const, 18446744073709551615ul>, luisa::compute::cuda::CUDAShaderMetadata const&, eastl::vector<luisa::compute::Argument, eastl::allocator>) + 1546
     3 [0x7fd9eb4c9ad3]: /home/kaspar/repos/Genesis/genesis/ext/LuisaRender/build/bin/liblc-backend-cuda.so :: luisa::compute::cuda::CUDADevice::create_shader(luisa::compute::ShaderOption const&, luisa::compute::Function) + 3443
     4 [0x7fb9b86e4dd6]: /home/kaspar/repos/Genesis/genesis/ext/LuisaRender/build/bin/libluisa-render-film-color.so :: unknown + 163286
     5 [0x7fd9f4161486]: /home/kaspar/repos/Genesis/genesis/ext/LuisaRender/build/bin/libluisa-render-util.so :: std::thread::_State_impl<std::thread::_Invoker<std::tuple<luisa::render::ThreadPool::Impl::Impl(unsigned long)::{lambda()#1}> > >::_M_run() + 214
     6 [0x7fdbd2bc6793]: /lib/x86_64-linux-gnu/libstdc++.so.6 :: unknown + 944019
     7 [0x7fdbd7e81609]: /lib/x86_64-linux-gnu/libpthread.so.0 :: unknown + 34313
     8 [0x7fdbd7fbb353]: /lib/x86_64-linux-gnu/libc.so.6 :: clone + 67
```

Research into this error (admittedly using GPT 4o a lot) strongly suggested this being due to a mismatch between CUDA driver and toolkit version and my GPU's compute capabilities. I am using CUDA toolkit 12.6, CUDA driver 565.57.01 and my GPU is NVIDIA GeForce GTX 1080 Ti which has compute capability 6.1. They should be compatible, afaic. 

So my next hunch was that rendering a .obj file that uses a material - which mine does - needs, NVTT to be installed, so I downloaded NVTT tarball for LINUX x86_64 from [NVidia's official website](https://developer.nvidia.com/gpu-accelerated-texture-compression) and moved all the binaries to `/usr/local/bin` and the shared object file to `/usr/local` and the libs to `/usr/local/lib`, and set `NVTT_DIR=/usr/local` and tried to re-compile LuisaRenderPy with 
```
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D CMAKE_CUDA_COMPILER=/usr/local/cuda-12.6/bin/nvcc and -D LUISA_COMPUTE_ENABLE_CUDA=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF -DCMAKE_CUDA_ARCHITECTURES=61 -D NVTT_DIR=/usr/local
cmake --build build -j4
```
but that fails with 
```
[ 66%] Building CXX object src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/extensions/cuda_texture_compression.cpp.o                                                                                                                                                                                                                                                                                            
In file included from /usr/include/X11/Xlib.h:44,                                                                                                                                                                                                                                                                                                                                                                                       
[... cut ...]                                                      
make: *** [Makefile:136: all] Error 2                                                                                          
```

My OS is Ubuntu 20.04. I have tried modifying LuisaRenderPy's C++ code even though I'm not particularly familiar with C++, but I'm sure there should be a way to compile LuisaRenderPy with NVTT without changing its code. The commit of LuisaRenderPy I have checked out is `eafb470e97cbc36368f610afa440bb03c4ee3703`.

Kashu7100, 2025-01-03:
Could you try 12.4 or 12.1 instead of 12.6?

kaspar-gestalt, 2025-01-04:
@Kashu7100 Thanks for the reply. I tried with CUDA 12.4 and it didn't change anything. Only when I downgraded the CUDA drivers from 565 to 550 did it work.

Kashu7100, 2025-01-04:
Good to hear that. My understanding is 12.4 - 550, 12.1 - 535 is the right combination.

kaspar-gestalt, 2025-01-04:
I should also mention that I inserted `#unset Success` after a bunch of Xlib imports to get around the error posted above. Then, an error in creating a shader occurred, which was solved by installing a different CUDA driver version.

closed 2025-01-10 by Kashu7100