Genesis-Embodied-AI/genesis-world #466: visualizer lottery 3 out of 5 - NOBODY Viewer-lucky on the Mac so far

ai-bits, 2025-01-03:
I went all the way back thru the open issues and I find it pretty annoying when you act as if it worked on the Mac and among half a dozen issues on that, **NOBODY** seems to have been able to see the Viewer.

I just did a
`git pull`
`pip install -U -e .` #to update Genesis Python 3.12.8 env

It says it updated Genesis to 0.2.1 and taichi to 1.7.3, but running `render_on_macos.py` **I still don't see no nothin**.
`<arrrrgghh>`
If some dev found it working you can't claim it does. Looks like you didn't try any new installation on any non-vanilla x64 + Nvidia machine on Linux like my Xeon + RTX 4000 Ada (or CPU-only?).

As stated in my **'visualizer lottery' issue series**, even omni-present x64 Intel+Nvidia GPU laptops on Linux need special treatment / knowledge (force Nvidia) to get the Viewer working.

Regards
G.

zhouxian, 2025-01-03:
Apologies for it. We tested internally on various macos chips and they all worked...
We are kinda short in manpower now, but we will conduct a thorough investigation across all hardware settings

ai-bits, 2025-01-03:
> We tested internally on various macos chips and they all worked...

What I suspected. BAD idea to launch an alpha 4 workdays before Xmas holidays.
Where is any Nvidia expert or any volunteer from the dozen of universities quoted in the docs?

As the `render_on_mac.py` silently runs through on my M3 MBP just **not** showing the viewer, I'm resorting to the output of @gmmyung 's [Visualizer does not work on macOS](https://github.com/Genesis-Embodied-AI/Genesis/issues/415) issue to get the idea what's going on behind the scenes.

At the first glance I see that `render_on_mac.py` expects an arg for `scene.viewer.start()`.
As it's getting rrreally late again: Before I begin to analyze, I'd like to ask you if you have any idea what's expected?

Thanks
G.

ai-bits, 2025-01-04:
Hey @zhouxian , Xmas & New Year holidays worker hero,
github says you are the author of `render_on_macos.py`, but I guess things are simply published by you / under your name!?

Anyway. So far nobody seems to have looked at the code, because at the first glance they would have seen that they need to provide the `-v` argument to really start the viewer.

Question remains, **why the heck would a programmer use the default `False` for `scene.viewer.start()` in code called `render_on_macos.py`???**

`parser.add_argument("-v", "--vis", action="store_true", default=False)`
...
`gs.tools.run_in_another_thread(fn=run_sim, args=(scene, args.vis))`
`if args.vis:`
`    scene.viewer.start()`

Sadly I don't remember where I have the pointer to `render_on_macos.py` in the repo, because there is nothing in the docs and in the readme, **so a default `True` to start the viewer in`render_on_macos.py` would have been REALLY HELPFUL.**
Cheers
G.

neil-tan, 2025-02-11:
@zhouxian @YilingQiao 
It seems people are having trouble with pip/pyenv setup on Apple Silicon.
I noticed that unless python is built with Tk support,  pyrender would fail to import from `ext. pyrender`. The symptom manifest as `trimesh` not defined as its import proceeded immediately after pyrender's.

I've put together some instructions on how to get around this: http://github.com/neil-tan/genRL
Please note that my setup.py is explicitly installing pyrender which leads to "GL invalid operation" like [here](https://github.com/Genesis-Embodied-AI/Genesis/issues/93). One may choose to comment pyrender out of the setup.py, but things wouldn't get as far.

duburcqa, 2025-02-25:
@ai-bits I have been working on fixing the various rendering issues lately. Can you summarise was is still not working at this time, as the discussion is going in many directions. From what I understand, 
* [ ] hybrid Intel + Nvidia GPU with On-Demand PRIME mode on Linux

duburcqa, 2025-02-25:
https://github.com/Genesis-Embodied-AI/Genesis/pull/782  introduces Interactive rendering on Mac OS.

duburcqa, 2025-02-27:
https://github.com/Genesis-Embodied-AI/Genesis/pull/796 fixes cameras on Windows OS.

duburcqa, 2025-02-27:
https://github.com/Genesis-Embodied-AI/Genesis/pull/798 brings offscreen rendering on Windows OS.

duburcqa, 2025-02-28:
https://github.com/Genesis-Embodied-AI/Genesis/pull/810 + https://github.com/Genesis-Embodied-AI/Genesis/pull/809 may have fixed rendering through VMware on Linux OS and Hybrid Intel + Nvidia GPU with On-Demand PRIME mode on Linux OS. To be confirmed.

closed 2025-05-24 by duburcqa