Genesis-Embodied-AI/genesis-world #468: Feature Request: Community-driven Development of Muscle and Tendon Models in Genesis

Hippozhibos, 2025-01-04:
### Background
In the Genesis open-source community, many users have expressed the desire to simulate muscle and tendon models within the existing framework (as discussed in Issues #374 and #287). However, these features are currently not supported, and due to limited resources, the developers may not be able to implement these features in the short term.

### Proposal
Would it be possible to organize a community-driven development team to develop muscle and tendon models in Genesis? Through this development, we hope to expand the range of applications for Genesis, particularly in the field of biomechanics. The specific steps would be as follows:
1. **Form a Community Team**: Invite developers interested in muscle and tendon modeling to join and collaborate on the development of this feature.
2. **Progressive Implementation**: Start with basic features (such as support for fixed tendons) and gradually expand to more complex muscle simulation (e.g., activation-contraction dynamics).
3. **Documentation and Tutorials**: After the functionality is developed, provide detailed documentation and use cases to help community users quickly get started.

### Requests and Suggestions
In order to ensure the smooth implementation of this proposal, I would like to ask the team the following questions:
1. **Technical Direction**:
   - Do you have any technical plans or implementation ideas for muscle and tendon simulations?
   - If we were to extend these features based on the current codebase, are there any recommended modules or pathways to follow?
2. **Community Collaboration**:
   - Do you support this community-driven development approach?
   - Would it be possible to open a dedicated discussion area on GitHub for this plan?
3. **Resources and Support**:
   - If the community team encounters technical difficulties during development, would it be possible to receive support from the official team?
   - Could this plan be promoted through official channels to attract more developers to participate?

Looking forward to the team's feedback. Thank you for your hard work and support for the open-source community!

Best regards

hudsonburke, 2025-01-04:
I would definitely love to be a part of this if it seems feasible! I work a lot with OpenSim for biomechanics, and would love to be able to integrate it with Genesis.

zhouxian, 2025-01-04:
Thank you so much for your interest in developing this!

tendon support is in our roadmap but we haven’t started it. To give you a brief idea of the current status of the team: we are on a new year break and I personally is also busy with a detailed speed benchmarking report on Genesis

we will come back with full thrust and an engineering team starting working on these features around late Jan

we will also release the benchmark report in two days, and also open a discord channel to facilitate discussions like this

thank you again on this! Given that we haven’t started, it’s probably a good idea if there’s contribution from the community and we can take it from there

Hippozhibos, 2025-01-04:
> Thank you so much for your interest in developing this!
>
> tendon support is in our roadmap but we haven’t started it. To give you a brief idea of the current status of the team: we are on a new year [...]

Thank you for your response and for sharing the exciting updates! I’m glad to hear tendon support is in the roadmap, and I look forward to the benchmark report and the upcoming Discord channel for discussions.

While waiting for the team to fully engage, I plan to continue exploring Genesis, especially in the area of muscle and tendon modeling, and will be happy to contribute where I can. If there are any specific tasks or directions I can start working on in the meantime, I’d be glad to help.

Another question: when tendon support is available, what file format would you recommend for defining musculoskeletal models? Should I use XML, MJCF, or URDF, and which would integrate best with the tendon model?

Looking forward to collaborating further and contributing to the project!

zhouxian, 2025-01-04:
I think MJCF would be good as i don't think urdf supports tendon?

Vittorio-Caggiano, 2025-01-04:
if you decide to go via MJCF, it would be great to use [models](https://github.com/MyoHub/myo_sim) and [tasks](https://github.com/MyoHub/myosuite) from MyoSuite. In this way we would be able to benchmark directly them against MuJoCo. Happy to help to test those and to benchmark.

Hippozhibos, 2025-01-06:
> I think MJCF would be good as i don't think urdf supports tendon?

Thanks for the suggestion! I’ll dive into MJCF and look forward to tendon support when it's ready!

Hippozhibos, 2025-01-06:
> if you decide to go via MJCF, it would be great to use [models](https://github.com/MyoHub/myo_sim) and [tasks](https://github.com/MyoHub/myosuite) from MyoSuite. In this way we would be able to benchmark directly them against MuJoCo. Happy to help to test those and to benchmark.

Thanks for the suggestion! Unfortunately, my focus is not human's musculoskeletal models, so MyoSuite won’t be applicable in my case. But I appreciate the offer and will keep it in mind!

Hippozhibos, 2025-02-07:
Hey @zhouxian and @Kashu7100 ,

Just checking in on the tendon modeling progress. Any updates on the timeline of tendon or muscle? Looking forward to collaborating and contributing as things move forward!

duburcqa, 2025-03-14:
@ale-amic1 I think it should not do that despite the lack of support of tendons and adhesion actuators... I think it may be related to collision exceptions not being supported either.