Genesis-Embodied-AI/genesis-world #470: Feature Request: Add Option to Control Mesh Detail Level in URDF Loading

NowLoadY, 2025-01-05:
Hello Genesis team,

When loading URDF files, I noticed that the mesh details appear to be significantly reduced compared to the original STL files. While this is great for performance, I need to preserve the original mesh details in some cases.

I see that the Mesh class has `decimate` and `decimate_face_num` parameters, but these cannot be controlled when loading through the URDF class. Would it be possible to expose these parameters in the URDF class as well?

Also, I'd appreciate any suggestions for alternative approaches to preserve mesh details when loading URDF files. Is there perhaps another way to achieve this with the current API?

Thank you for your help!

Best regards
[image]

zhouxian, 2025-01-05:
In general simulation it’s recommended that each collision mesh only has a few hundred vertices max, while visualization mesh can be much more complex

yes we could definitely add an API

could you update with some more visuals with a better view? The mesh you are importing and how it looks like now

NowLoadY, 2025-01-05:
Thank you for the quick response! I understand the recommendation about separating collision and visualization meshes. 

I have a potential workaround in mind - would it be possible to:
1. Load the URDF with `visualization=False` for physics simulation
2. Then add another entity using `gs.morphs.Mesh` at the same position with `decimate=False` for visualization only

This way we could keep the efficient collision meshes from URDF while having high-detail visualization. Would this approach work with the current API? Or do you see any potential issues with this solution?
[image]

zhouxian, 2025-01-05:
Ah no i think you misunderstood -- each entity has both visual geoms and collision geoms, and we use visual geoms for rendering. If you use `vis_mode='collision'`, then genesis will render the underlying collision geoms

In your case, what does the original mesh look like? The visualization mesh should be identical in Genesis -- we didn't do any preprocessing to simplify it

closed 2025-03-08 by YilingQiao