Genesis-Embodied-AI/genesis-world #471: Different Entities pos in Parallel Simulation Environments?  Put camera on each robot?

kanghui0204, 2025-01-05:
Hi Genesis experts,

Thank you very much for your framework. I am currently planning to use the framework to train some RL models, and I have a few questions that I would like to get answered. I noticed that there is an issue(https://github.com/Genesis-Embodied-AI/Genesis/issues/464) here that is quite similar to mine, but I would still like to double-check. 

**My Requirements**
I would like to train a navigation task for an ANYmal robot. In the navigation task, I need to include some obstacles (you can think of them as one or more cubes) and a camera mounted on the front of the robot. The positions of these obstacles, the navigation task's endpoint, and the robot's starting position are all random values.

**Questions**

1. Is it possible to have N cubes in each env with different positions? N is fixed， every env have N cubes , only position is different.
2. How can I attach a camera to each robot so that I can obtain camera images in the observations and train a CNN+MLP RL model?

Thank you very much for your assistance.

zhouxian, 2025-01-05:
Yes you can add N cubes, and set them to be in different positions. i think `entity.set_position` will do the job

we don't have APIs for attaching camera yet -- you will for now have to manually update camera pose each step... but we should be able to add one very soon

kanghui0204, 2025-01-05:
Thank you for your quick response , you mean ,when I call a reset function like below :
```
def reset(self): 
    self.reset_buf[:] = True
    self.reset_idx(torch.arange(self.num_envs, device=self.device)) 
    return self.obs_buf, None  
```

I can add N cube.set_position(tensor) ,which tensor is [num_envs , postion_dim] ,so every env's cude have different position?

zhouxian, 2025-01-05:
Yes

kanghui0204, 2025-01-06:
OK thank you for your confirm , I will close this issue

closed 2025-01-06 by kanghui0204

wangyian-me, 2025-01-22:
We've updated a function about how to attach or detach a camera on a rigid link (of an articulated object or a mesh object). 
Below is one example usage:
```
...
cam = scene.add_camera(
    pos=(3, -1, 1.5),
    lookat=(0.0, 0.0, 0.5),
    res=(1024, 1024),
    fov=30,
    GUI=False,
)
...
franka = scene.add_entity(
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml")
)
...
end_effector = franka.get_link("hand")
offset_T = np.array([
    [1,  0,  0,  0.05],
    [0,  1,  0,  0.05],
    [0,  0,  -1,  -0.05],
    [0,  0,  0,  1.0]
])
cam.attach(rigid_link=end_effector, offset_T=offset_T)
...
```
By this code, you can attach the camera to a rigid link with a predefined offset pose matrix (adjust the matrix if needed), and the camera will move along with the rigid link while keeping this pose offset.

**Note that this function is not compatible with parallel simulation yet, since that might result in different camera pose in different parallel environment, which is not supported yet.**

vlamirand, 2025-03-25:
Hi! Have you already added support for parallel cameras in the simulation? I need to know quickly.

mhyatt000, 2025-04-18:
@wangyian-me 
>Note that this function is not compatible with parallel simulation yet, since that might result in different camera pose in different parallel environment, which is not supported yet.

Is there a timeline for supporting this? I have been thinking about switching to Genesis, but use eye-in-hand cameras for manipulation

anish-bhattacharya, 2025-05-02:
Not meaning to provide any pressure, but I would similarly like to emphasize the impact of enabling parallel environments with rigidly-attached cameras / independently moving cameras (point #1 by mhyatt000). This would be a huge boon to the vision-based robotics field :)