Genesis-Embodied-AI/genesis-world #473: BUG. gs.inv_transform_by_trans_quat return different shape given numpy array and torch tensor.

Riften, 2025-01-05:
As described in the title.

```python
print(gs.inv_transform_by_trans_quat(
    torch.as_tensor([1,2,3], dtype=gs.tc_float, device=gs.device),
    torch.as_tensor([4,5,6], dtype=gs.tc_float, device=gs.device),
    torch.as_tensor([1,0,0,0], dtype=gs.tc_float, device=gs.device)
).shape)
print(gs.inv_transform_by_trans_quat(
    np.array([1,2,3]), 
    np.array([4,5,6]), 
    np.array([1,0,0,0])).shape)
```

output
```bash
torch.Size([1, 3])
(3,)
```

Riften, 2025-01-06:
Is this a bug or a feature?

zhenjia-xu, 2025-01-09:
It is a bug. Thanks for pointing it out.
It has been fixed in #518. Let me know if it solves your issue.

Riften, 2025-01-09:
> It is a bug. Thanks for pointing it out. It has been fixed in #518. Let me know if it solves your issue.

It is solved. Thank you!

closed 2025-01-09 by Riften