Genesis-Embodied-AI/genesis-world #474: No display detected

ZhihongLiao, 2025-01-05:
(genesis) (base) liaozh@ubuntu-Super-Server:~/Genesis$ /home/liaozh/anaconda3/envs/genesis/bin/python /home/liaozh/Genesis/examples/tutorials/hello_genesis.py
[Genesis] [21:24:50] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [21:24:50] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [21:24:50] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [21:24:50] [INFO] Running on [Intel(R) Xeon(R) Platinum 8336C CPU @ 2.30GHz] with backend gs.cpu. Device memory: 251.30 GB.
[Genesis] [21:24:50] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/home/liaozh/Genesis/examples/tutorials/hello_genesis.py", line 5, in <module>
    scene = gs.Scene()
  File "/home/liaozh/Genesis/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/home/liaozh/Genesis/genesis/engine/scene.py", line 147, in __init__
    self._visualizer = Visualizer(
  File "/home/liaozh/Genesis/genesis/vis/visualizer.py", line 37, in __init__
    gs.raise_exception("No display detected. Use `show_viewer=False` for headless mode.")

[Genesis] [21:24:51] [ERROR] No display detected. Use `show_viewer=False` for headless mode.
[Genesis] [21:24:51] [INFO] 💤 Exiting Genesis and caching compiled kernels...

zswang666, 2025-01-08:
You need display. You can either (i) run with a physical machine with a monitor instead of ssh, (ii) set `show_viewer=False` as suggested in the error message, or (iii) do `export DISPLAY=:?` where the number `?` should be the display from your machine 

Also please check out related issues https://github.com/Genesis-Embodied-AI/Genesis/issues?q=is%3Aissue+display

duburcqa, 2025-03-22:
Nothing actionable on Genesis side.

closed 2025-03-22 by duburcqa