Genesis-Embodied-AI/genesis-world #475: Importing custom urdf file

YunSang123, 2025-01-05:
Hi, I am a beginner to Genesis. In my knowledge, there isn't any tutorial page for a beginner for me to import custom urdf file.

How can I import my custom urdf file?

Also, I want to import my .urdf.xacro file. Is it currently impossible to load any urdf.xacro file?

Thanks!

zhouxian, 2025-01-05:
How exactly is your custom urdf different? You should be able to load it
just like how we load other urdf in the tutorial

YunSang123, 2025-01-05:
Oh, I forgot the stl file. Thanks!

Also, I was wondering whether genesis can import .urdf.xacro file.

zhouxian, 2025-01-06:
I don’t think so. Not yet

could you briefly describe how is it different than regular urdf?

oscar-lima, 2025-01-06:
From http://wiki.ros.org/xacro and https://github.com/ros/xacro:

"Xacro is an XML macro language. With xacro, you can construct shorter and more readable XML files by using macros that expand to larger XML expressions."

So you can create "functions" (called macros) and "variables" (called properties), include other xacro files and resolve absolute xacro paths from ROS packages using functions such as "find package" among others. 

One caveat is that xacro relies heavily on ROS. An alternative would be to borrow [xacro_standalone.py](https://github.com/bulletphysics/bullet3/blob/master/data/xacro_standalone.py) from [pybullet](https://github.com/bulletphysics/bullet3/tree/master).

NOTE: [ROS](https://ros.org/) users can generate (more like compile...) URDF from xacro easily (with one command) and save the result locally. This can be then loaded into genesis. However is annoying to generate the file over and over again if you make changes or for users than don't (or don't want to) have ROS installed.