Genesis-Embodied-AI/genesis-world #480: How to get shadow with ray tracing

Progolue, 2025-01-06:
[image]
I can't get shadow with ray tracing,did i do something wrong?
`        renderer=gs.renderers.RayTracer(
            env_surface=gs.surfaces.Emission(
                emissive_texture=gs.textures.ImageTexture(
                    image_path="textures/indoor_bright.png",
                ),
            ),
            env_radius=5.0,
            env_pos = (0,0,20),
            env_euler=(0, 0, 180),
            # lights=config['scene']['lights'],
            lights=[
                    {"pos": (0.0, 0.0, 10.0), "color": (2.0, 2.0, 2.0), "intensity": 10.0, "radius": 3.0},
            ],
        ),
    ) `Thank you for answering me

zhouxian, 2025-01-06:
Try removing the env surface?

zhouxian, 2025-01-06:
I think the shadow is there — it’s just not visible because of the lighting condition

The latter should be a bug. At some point we renamed id to uid and probably missed that…

Progolue, 2025-01-06:
> I think the shadow is there — it’s just not visible because of the lighting condition
> 
> The latter should be a bug. At some point we renamed id to uid and probably missed that…
Thank you for your reply. Since I’m currently unable to adjust the light angle through the add_light method due to this issue, do you have any other suggestions or methods I could use to customize the light direction? Any guidance would be greatly appreciated.

zhouxian, 2025-01-06:
You can add any entity and specify the surface to be a emissive surface so that it functions as a light

we will add a tutorial soon

Kashu7100, 2025-01-06:
@Progolue can you give me the full error msg? I can check for the bug.

Progolue, 2025-01-06:
> @Progolue can you give me the full error msg? I can check for the bug.
Thank you for helping, here is the full error message:
Traceback (most recent call last):
  File "/home/huang/modeltool/Genesis/examples/rendering/autorender/dataproc.py", line 184, in <module>
    syn_data()
  File "/home/huang/modeltool/Genesis/examples/rendering/autorender/dataproc.py", line 87, in syn_data
    light = scene.add_light(
  File "/home/huang/modeltool/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/huang/modeltool/Genesis/genesis/engine/scene.py", line 442, in add_light
    self.visualizer.raytracer.add_mesh_light(
  File "/home/huang/modeltool/Genesis/genesis/vis/raytracer.py", line 200, in add_mesh_light
    name=str(mesh._id),
AttributeError: 'Mesh' object has no attribute '_id'

Progolue, 2025-01-06:
> You can add any entity and specify the surface to be a emissive surface so that it functions as a light
> 
> we will add a tutorial soon

Thank you for the information! I'll look forward to the tutorial.

closed 2025-01-08 by zhouxian