Genesis-Embodied-AI/genesis-world #481: ImportError: /mujoco/_functions.cpython-39-x86_64-linux-gnu.so: undefined symbol: mju_sparse2dense

hiroki-hashimoto-24, 2025-01-06:
**Describe the bug**
I'm trying to use Genesis but I can't solve the import error, what should I do?

**To Reproduce**
Steps to reproduce the behavior:
python3 Genesis/examples/tutorials/hello_genesis.py

**Expected behavior**
ImportError: /home/hashimoto/.local/lib/python3.10/site-packages/mujoco/_functions.cpython-310-x86_64-linux-gnu.so: undefined symbol: mju_sparse2dense

**Screenshots**
[image]

**Environment**
CPU/GPU: [Intel(R) Core(TM) i7-7700K/NVIDIA GeForce GTX 1080 Ti]
OS: [Ubuntu 22.04]
ROS: [ROS2 humble]
CUDA: [CUDA 11.8]

zhouxian, 2025-01-06:
Which municipal version are u using?

hiroki-hashimoto-24, 2025-01-06:
My MuJoCo version is 3.2.7.

duburcqa, 2025-03-23:
I recommend updating to the latest release of mujoco, which is now officially by Genesis. There is no much that can be done on Genesis side beyond that. Closing.

closed 2025-03-23 by duburcqa