Genesis-Embodied-AI/genesis-world #486: WSL Build LuisaRender got [Makefile:136: all] Error 2

zhangiser, 2025-01-06:
My env:
```bash
WSL:2
CUDA:12.4.2
Ubuntu:22.04
conda:24.11.1
Pytorch:2.5.1
NVIDIA Driver：566.36
Python：3.9.21
```
First i run success with `cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF. `  Despite several warnings, it was still successful.

Then i run failed with `cmake --build build -j $(nproc)`
```bash
(torch) root@zhangiser-pc:/usr/Genesis/genesis/ext/LuisaRender# cmake --build build -j $(nproc)
[  0%] Building C object src/compute/src/ext/reproc/reproc/CMakeFiles/reproc.dir/src/clock.posix.c.o

..............................................................................

[ 65%] Building CXX object src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/extensions/cuda_texture_compression.cpp.o
In file included from /usr/include/X11/Xlib.h:44,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/../../common/vulkan_instance.h:20,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/../cuda_event.h:8,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.cpp:11:
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:21:9: error: expected identifier before numeric constant
   21 |         Success = 0,
      |         ^~~~~~~
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:21:9: error: expected ‘}’ before numeric constant
In file included from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:4,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.cpp:12:
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:19:32: note: to match this ‘{’
   19 |     enum class Result : int8_t {
      |                                ^
In file included from /usr/include/X11/Xlib.h:44,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/../../common/vulkan_instance.h:20,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/../cuda_event.h:8,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.cpp:11:
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:21:9: error: expected unqualified-id before numeric constant
   21 |         Success = 0,
      |         ^~~~~~~
In file included from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:4,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.cpp:12:
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:25:13: error: ‘Result’ does not name a type; did you mean ‘VkResult’?
   25 |     virtual Result compress_bc6h(Stream &stream,
      |             ^~~~~~
      |             VkResult
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:28:13: error: ‘Result’ does not name a type; did you mean ‘VkResult’?
   28 |     virtual Result compress_bc7(Stream &stream,
      |             ^~~~~~
      |             VkResult
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:32:13: error: ‘Result’ does not name a type; did you mean ‘VkResult’?
   32 |     virtual Result check_builtin_shader() noexcept { return Result::NotImplemented; }
      |             ^~~~~~
      |             VkResult
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:35:1: error: expected declaration before ‘}’ token
   35 | }// namespace luisa::compute
      | ^
In file included from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.cpp:12:
/usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:28:12: error: luisa::compute::TexCompressExt::Result luisa::compute::cuda::CUDATexCompressExt::compress_bc6h(luisa::compute::Stream&, const luisa::compute::ImageView<float>&, const luisa::compute::BufferView<unsigned int>&)’ marked ‘override’, but does not override
   28 |     Result compress_bc6h(Stream &stream,
      |            ^~~~~~~~~~~~~
/usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:31:12: error: luisa::compute::TexCompressExt::Result luisa::compute::cuda::CUDATexCompressExt::compress_bc7(luisa::compute::Stream&, const luisa::compute::ImageView<float>&, const luisa::compute::BufferView<unsigned int>&, float)’ marked ‘override’, but does not override
   31 |     Result compress_bc7(Stream &stream,
      |            ^~~~~~~~~~~~
/usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:35:12: error: luisa::compute::TexCompressExt::Result luisa::compute::cuda::CUDATexCompressExt::check_builtin_shader()’ marked ‘overrid’, but does not override
   35 |     Result check_builtin_shader() noexcept override;
      |            ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/X11/Xlib.h:44,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/../../common/vulkan_instance.h:20,
                 from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/../cuda_event.h:8,
[... 65 lines cut ...]
      |             ^~~~~~
      |             VkResult
/usr/Genesis/genesis/ext/LuisaRender/src/compute/include/luisa/backends/ext/tex_compress_ext.h:35:1: error: expected declaration before ‘}’ token
   35 | }// namespace luisa::compute
      | ^
In file included from /usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/cuda_device.cpp:48:
/usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:28:12: error: luisa::compute::TexCompressExt::Result luisa::compute::cuda::CUDATexCompressExt::compress_bc6h(luisa::compute::Stream&, const luisa::compute::ImageView<float>&, const luisa::compute::BufferView<unsigned int>&)’ marked ‘override’, but does not override
   28 |     Result compress_bc6h(Stream &stream,
      |            ^~~~~~~~~~~~~
/usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:31:12: error: luisa::compute::TexCompressExt::Result luisa::compute::cuda::CUDATexCompressExt::compress_bc7(luisa::compute::Stream&, const luisa::compute::ImageView<float>&, const luisa::compute::BufferView<unsigned int>&, float)’ marked ‘override’, but does not override
   31 |     Result compress_bc7(Stream &stream,
      |            ^~~~~~~~~~~~
/usr/Genesis/genesis/ext/LuisaRender/src/compute/src/backends/cuda/extensions/cuda_texture_compression.h:35:12: error: luisa::compute::TexCompressExt::Result luisa::compute::cuda::CUDATexCompressExt::check_builtin_shader()’ marked ‘overrid’, but does not override
   35 |     Result check_builtin_shader() noexcept override;
      |            ^~~~~~~~~~~~~~~~~~~~
[ 65%] Linking CXX shared module ../../../../../bin/liblc-backend-remote.so
[ 65%] Linking CXX shared module ../../../../../bin/liblc-backend-cpu.so
[ 65%] Built target luisa-compute-backend-remote
[ 65%] Built target luisa-compute-backend-cpu
gmake[2]: *** [src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/build.make:544: src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/extensions/cuda_texture_compression.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/build.make:144: src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/cuda_device.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2544: src/compute/src/backends/cuda/CMakeFiles/luisa-compute-backend-cuda.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

Kashu7100, 2025-01-07:
Could you try Docker build for sanity check?

rosefirefish, 2025-01-08:
I got exactly same error with environment:
```
Ubuntu 22.04
CUDA:12.1.66
conda:23.3.1
Pytorch:2.5.1
NVIDIA Driver：555.42
Python：3.11.11
cmake: 3.31.3
gcc: 11.4.0
g++: 11.4.0
```
I tried to run with docker, it did NOT stop here, but it crashed at 73% of the building process twice.

I tried to run the code in docker file too. Namely, apart from the base docker image, I ran the remaining codes and my cuda env is setup globally, I got the same error in my env.

Here is nvida-smi result
```  
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 555.42.06              Driver Version: 555.42.06      CUDA Version: 12.5     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4090        Off |   00000000:01:00.0  On |                  Off |
|  0%   46C    P8             37W /  450W |     640MiB /  24564MiB |     23%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+
```

rosefirefish, 2025-01-09:
This issue can be solved by downgrading the driver to a version compatible to CUDA (12.1 - 535, 12.4 - 550 according to what mentioned in the answer [465](https://github.com/Genesis-Embodied-AI/Genesis/issues/465#issuecomment-2570018322). I tried the 12.1 -  535 one.
@Kashu7100 Maybe such info can be added to the instruction? (As we normally assume that as long as the CUDA version is no higher than what the driver supports, it should be fine.)

Also, if anyone find a similar crash problem as me, increase the size of the swap partition, or, add `-j1` when building...

closed 2025-03-06 by YilingQiao