Genesis-Embodied-AI/genesis-world #492: Segmentation fault (core dumped)

HaodongHuang-W, 2025-01-07:
I am training locomotion in Genesis, and once the number of training iterations exceeds 1000, there may be some egmentation fault (core dumped) . May I ask if this is a computer problem or something else? 
[image]

zhouxian, 2025-01-07:
Hmm  core dump sometimes happens even when you exit normally so it’s not really saying anything

could you double check if there’s anything in you code that stopped the execution from your side?

HaodongHuang-W, 2025-01-13:
@zhouxian
I add 
`import faulthandler
faulthandler.enable()` 
The following error occurred:
[image]

HaodongHuang-W, 2025-01-13:
The specific errors are as follows：
Fatal Python error: Segmentation fault

Thread 0x00007f9fb1cff700 (most recent call first):

Thread 0x00007fa040e2b700 (most recent call first):
File "/home/ubuntu/anaconda3/envs/genesis/lib/python3.10/threading.py", line 324 in wait
File "/home/ubuntu/anaconda3/envs/genesis/lib/python3.10/queue.py", line 180 in get
File "/home/ubuntu/anaconda3/envs/genesis/lib/python3.10/site-packages/tensorboard/summary/writer/event_file_writer.py", line 269 in _run
File "/home/ubuntu/anaconda3/envs/genesis/lib/python3.10/site-packages/tensorboard/summary/writer/event_file_writer.py", line 244 in run
File "/home/ubuntu/anaconda3/envs/genesis/lib/python3.10/threading.py", line 1016 in _bootstrap_inner
File "/home/ubuntu/anaconda3/envs/genesis/lib/python3.10/threading.py", line 973 in _bootstrap

Current thread 0x00007fa2972464c0 (most recent call first):
File "/home/ubuntu/Genesis/examples/locomotion/Genesis-backflip/reward_wrapper.py", line 88 in _reward_feet_air_time
File "/home/ubuntu/Genesis/examples/locomotion/Genesis-backflip/env.py", line 498 in compute_reward
File "/home/ubuntu/Genesis/examples/locomotion/Genesis-backflip/env.py", line 546 in post_physics_step
File "/home/ubuntu/Genesis/examples/locomotion/Genesis-backflip/env.py", line 754 in step
File "/home/ubuntu/Genesis/rsl_rl/rsl_rl/runners/on_policy_runner.py", line 108 in learn
File "/home/ubuntu/Genesis/examples/locomotion/Genesis-backflip/train_walk.py", line 254 in main
File "/home/ubuntu/Genesis/examples/locomotion/Genesis-backflip/train_walk.py", line 258 in

Extension modules: numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, torch._C, torch._C._dynamo.autograd_compiler, torch._C._dynamo.eval_frame, torch._C._dynamo.guards, torch._C._dynamo.utils, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special, psutil._psutil_linux, psutil._psutil_posix, PIL._imaging, scipy._lib._ccallback_c, scipy.sparse._sparsetools, _csparsetools, scipy.sparse._csparsetools, scipy.linalg._fblas, scipy.linalg._flapack, scipy.linalg.cython_lapack, scipy.linalg._cythonized_array_utils, scipy.linalg._solve_toeplitz, scipy.linalg._decomp_lu_cython, scipy.linalg._matfuncs_sqrtm_triu, scipy.linalg.cython_blas, scipy.linalg._matfuncs_expm, scipy.linalg._decomp_update, scipy.sparse.linalg._dsolve._superlu, scipy.sparse.linalg._eigen.arpack._arpack, scipy.sparse.linalg._propack._spropack, scipy.sparse.linalg._propack._dpropack, scipy.sparse.linalg._propack._cpropack, scipy.sparse.linalg._propack._zpropack, scipy.sparse.csgraph._tools, scipy.sparse.csgraph._shortest_path, scipy.sparse.csgraph._traversal, scipy.sparse.csgraph._min_spanning_tree, scipy.sparse.csgraph._flow, scipy.sparse.csgraph._matching, scipy.sparse.csgraph._reordering, scipy.spatial._ckdtree, scipy._lib.messagestream, scipy.spatial._qhull, scipy.spatial._voronoi, scipy.spatial._distance_wrap, scipy.spatial._hausdorff, scipy.special._ufuncs_cxx, scipy.special._ufuncs, scipy.special._specfun, scipy.special._comb, scipy.special._ellip_harm_2, scipy.spatial.transform._rotation, vtkmodules.vtkCommonCore, vtkmodules.vtkCommonMath, vtkmodules.vtkCommonTransforms, vtkmodules.vtkCommonDataModel, vtkmodules.vtkCommonExecutionModel, vtkmodules.vtkImagingSources, vtkmodules.vtkCommonMisc, vtkmodules.vtkFiltersCore, vtkmodules.vtkFiltersPython, vtkmodules.vtkCommonComputationalGeometry, vtkmodules.vtkIOCore, vtkmodules.vtkIOLegacy, vtkmodules.vtkIOXMLParser, vtkmodules.vtkRenderingCore, vtkmodules.vtkDomainsChemistry, vtkmodules.vtkFiltersGeneral, vtkmodules.vtkFiltersExtraction, vtkmodules.vtkFiltersFlowPaths, vtkmodules.vtkFiltersGeometry, vtkmodules.vtkFiltersHybrid, vtkmodules.vtkFiltersModeling, vtkmodules.vtkFiltersHyperTree, vtkmodules.vtkFiltersSources, vtkmodules.vtkFiltersTexture, vtkmodules.vtkParallelCore, vtkmodules.vtkFiltersParallel, vtkmodules.vtkFiltersParallelDIY2, vtkmodules.vtkFiltersPoints, vtkmodules.vtkFiltersStatistics, vtkmodules.vtkFiltersVerdict, vtkmodules.vtkImagingCore, vtkmodules.vtkImagingGeneral, vtkmodules.vtkImagingHybrid, vtkmodules.vtkIOGeometry, vtkmodules.vtkIOXML, vtkmodules.vtkIOInfovis, vtkmodules.vtkIOPLY, vtkmodules.vtkImagingMorphological, vtkmodules.vtkRenderingContext2D, vtkmodules.vtkPythonContext2D, vtkmodules.vtkImagingFourier, vtkmodules.vtkIOParallelXML, vtkmodules.vtkRenderingFreeType, kiwisolver._cext, tetgen._tetgen, scipy.optimize._group_columns, scipy.optimize._trlib._trlib, scipy.optimize._lbfgsb, _moduleTNC, scipy.optimize._moduleTNC, scipy.optimize._cobyla, scipy.optimize._slsqp, scipy.optimize._minpack, scipy.optimize._lsq.givens_elimination, scipy.optimize._zeros, scipy.optimize._highs.cython.src._highs_wrapper, scipy.optimize._highs._highs_wrapper, scipy.optimize._highs.cython.src._highs_constants, scipy.optimize._highs._highs_constants, scipy.linalg._interpolative, scipy.optimize._bglu_dense, scipy.optimize._lsap, scipy.optimize._direct, lxml._elementpath, lxml.etree, scipy.ndimage._nd_image, _ni_label, scipy.ndimage._ni_label, scipy.interpolate._fitpack, scipy.interpolate._dfitpack, scipy.interpolate._bspl, scipy.interpolate._ppoly, scipy.interpolate.interpnd, scipy.interpolate._rbfinterp_pythran, [... cut ...] ltersAMR, vtkmodules.vtkDomainsChemistryOpenGL2, vtkmodules.vtkCommonPython, vtkmodules.vtkChartsCore, vtkmodules.vtkAcceleratorsVTKmCore, vtkmodules.vtkAcceleratorsVTKmDataModel, vtkmodules.vtkAcceleratorsVTKmFilters, yaml._yaml, numba.core.typeconv._typeconv, numba._helperlib, numba._dynfunc, numba._dispatcher, numba.core.runtime._nrt_python, numba.np.ufunc._internal, numba.experimental.jitclass._box, PIL._imagingft, google._upb._message, PIL._imagingmath, lxml.builder (total: 251)
段错误 (核心已转储)

closed 2025-05-24 by duburcqa