Genesis-Embodied-AI/genesis-world #498: Height field terrain limits for large environment.

mi-robotics, 2025-01-07:
When attempting to load larger mesh terrains it results in a memory explosion.
Interestingly if you reduce this just slightly to a 320 by 320 heightfield the memory explosion does not occur.
```
horizontal_scale = 0.15
vertical_scale = 0.005
heights_range = np.arange(-10, 20, 10)
height_field = 0 + np.random.choice(heights_range, (350, 350))
terrain = self.scene.add_entity(
            morph=gs.morphs.Terrain(
                horizontal_scale=horizontal_scale,
                vertical_scale=vertical_scale,
                height_field=height_field,
            ),
)
```

The error provided
```
line 26: 21899 Killed: 9               python train_simple.py
```

I am unable to load this terrain on M1 mac with 32gb ram, nor a server (Ubuntu) with 40gb Vram.
In both cases the memory explodes, specifically this occurs during the Collider Intialization:
    /Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py
```
def _init_collider(self):
        self.collider = Collider(self)
```
 
Are there any good strategies to load larger trimesh/height field terrains, or some alternative that could enable training on varied curriculum based terrains. I think this will be required to compete with Isaac.

On a side note, I have found that on my Mac M1, the rigid/terrain_... tutorials fail at scene.add_entity()
```
OMP: Error #179: Function pthread_mutex_init failed:
OMP: System error #22: Invalid argument
Segmentation fault: 11
```

Would greatly appreciate any help.

Huisouan, 2025-01-08:
Have you tried loading terrain in an environment with env=4096? When Genesis builds scenes, it loads a terrain for each scene, which makes simulations requiring complex terrains, like the terrain course, impossible to load due to insufficient memory.

zhouxian, 2025-01-08:
It looks like something is definitely optimizable

in a internal version earlier we were able to create 30000 terrain ends as it’s just height fields so it should occupy too much memory

I’ll look into it

mi-robotics, 2025-01-08:
Hi @zhouxian thanks.

for more context my current changes are as follows:

Genesis/genesis/engine/solvers/rigid/support_field_decomp.py
in def _compute_support(self):

```
def _compute_support(self):
        v = self._get_direction_grid()
        v1 = v.reshape([-1, 3])
        support_v = []
        support_vid = []
        support_cell_start = []
        start = 0

        if self.solver.n_geoms > 0:

            init_pos = self.solver.verts_info.init_pos.to_numpy()
            gs.logger.info(f'iterator {self.solver.n_geoms}')
            for i_g in range(self.solver.n_geoms):

                vert_start = self.solver.geoms_info.vert_start[i_g]
                vert_end = self.solver.geoms_info.vert_end[i_g]
                this_pos = init_pos[vert_start:vert_end]

                # Split the computation into chunks
                chunk_size=10000
                max_indices = []
                max_values = []

                # Compute dot product iteratively to find max indices
                for start_idx in range(0, this_pos.shape[0], chunk_size):
                    end_idx = min(start_idx + chunk_size, this_pos.shape[0])
                    chunk = this_pos[start_idx:end_idx]

                    
                    # Compute dot product for the current chunk
                    dot_chunk = v1.dot(chunk.T)
                    
                    # Find local max indices and values for the chunk
                    local_max_indices = np.argmax(dot_chunk, axis=1)
                    local_max_values = dot_chunk[np.arange(dot_chunk.shape[0]), local_max_indices]
                    
                    # Adjust indices to global indices and store
                    max_indices.append(local_max_indices + start_idx)
                    max_values.append(local_max_values)

                # Combine results from all chunks
                max_indices = np.array(max_indices).flatten()
                support = this_pos[max_indices]

      
                support_cell_start.append(start)
                support_v.append(support)
                support_vid.append(max_indices)
                start += support.shape[0]

            support_v = np.concatenate(support_v)
            support_vid = np.concatenate(support_vid, dtype=gs.np_int)
            support_cell_start = np.array(support_cell_start, dtype=gs.np_int)
        else:
            support_v = np.zeros([1, 3], dtype=gs.np_float)
            support_vid = np.zeros([1], dtype=gs.np_int)
            support_cell_start = np.zeros([1], dtype=gs.np_int)

        self.n_support_cells = start
        self.support_cell_start = ti.field(dtype=gs.ti_int, shape=self.solver.n_geoms_)
        self.support_v = ti.Vector.field(3, dtype=gs.ti_float, shape=max(1, self.n_support_cells))
        self.support_vid = ti.field(dtype=gs.ti_int, shape=max(1, self.n_support_cells))
        self._kernel_init_support(support_cell_start, support_v, support_vid)
```

this solves large memory requirements for the dot product and argmax operations.
this works for large terrains and small number of envs 

now testing with large number of envs, i am getting int64 indexing issue in:
Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py
in  def _init_vert_fields(self) ad this part
```
self.verts_state = struct_vert_state.field(
            shape=self._batch_shape(self.n_verts_), needs_grad=False, layout=ti.Layout.SOA
        ) #size->(1019087, 4096))
```

I am looking to splitting the state into multiple int32 indexable items as this is a taichi limitation, but i'm not to sure on the down stream uses of vert_state, is this only used within the class do you know?

mi-robotics, 2025-01-08:
the int32 memory issue is fairly simple to address with:

Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py
```
class LargeTensor2DManager:
    def __init__(self, shape, max_chunk_elements):
        # Define struct
        struct_vert_state = ti.types.struct(pos=ti.math.vec3)

        self.total_rows, self.total_cols = shape

        # Calculate chunk dimensions
        self.chunk_rows = int(min(self.total_rows, max_chunk_elements ** 0.5))
        self.chunk_cols = int(min(self.total_cols, max_chunk_elements // self.chunk_rows))

        # Ensure chunk size does not exceed max_chunk_elements
        assert self.chunk_rows * self.chunk_cols <= max_chunk_elements

        self.num_row_chunks = (self.total_rows + self.chunk_rows - 1) // self.chunk_rows
        self.num_col_chunks = (self.total_cols + self.chunk_cols - 1) // self.chunk_cols

        # Create chunks
        self.chunks = []
        for r in range(self.num_row_chunks):
            row_chunks = []
            for c in range(self.num_col_chunks):
                current_chunk_shape = (
                    min(self.chunk_rows, self.total_rows - r * self.chunk_rows),
                    min(self.chunk_cols, self.total_cols - c * self.chunk_cols),
                )
                field = struct_vert_state.field(
                    shape=current_chunk_shape, needs_grad=False, layout=ti.Layout.SOA
                )
                row_chunks.append(field)
            self.chunks.append(row_chunks)

    def _get_chunk_indices(self, i, j):
        # Determine which chunk and local index within the chunk
        chunk_row_idx = i // self.chunk_rows
        chunk_col_idx = j // self.chunk_cols
        local_row_idx = i % self.chunk_rows
        local_col_idx = j % self.chunk_cols
        return chunk_row_idx, chunk_col_idx, local_row_idx, local_col_idx

    def __getitem__(self, global_idx):
        i, j = global_idx
        chunk_row_idx, chunk_col_idx, local_row_idx, local_col_idx = self._get_chunk_indices(i, j)
        return self.chunks[chunk_row_idx][chunk_col_idx][local_row_idx, local_col_idx]

    def __setitem__(self, global_idx, value):
        i, j = global_idx
        chunk_row_idx, chunk_col_idx, local_row_idx, local_col_idx = self._get_chunk_indices(i, j)
        self.chunks[chunk_row_idx][chunk_col_idx][local_row_idx, local_col_idx] = value
        
......
......
......

def _init_vert_fields(self):
        # collisioin geom

        struct_vert_info = ti.types.struct(
            init_pos=gs.ti_vec3,
            init_normal=gs.ti_vec3,
            geom_idx=gs.ti_int,
            init_center_pos=gs.ti_vec3,
        )

        struct_face_info = ti.types.struct(
            verts_idx=gs.ti_ivec3,
[... 402 lines cut ...]
              0.037 ms  0.12%  flag_access           [24 x   1.530 us]
              1.163 ms  3.74%  offload               [12 x  96.917 us]
                  0.010 ms  0.86%  replace_all_usages_with [12 x 834.465 ns]
                  0.104 ms  8.96%  type_check            [24 x   4.341 us]
                  1.049 ms 90.18%  [unaccounted]
              2.433 ms  7.83%  cfg_optimization      [12 x 202.755 us]
                  1.291 ms 53.05%  store_to_load_forwarding [12 x 107.567 us]
                      0.725 ms 56.15%  reaching_definition_analysis [12 x  60.399 us]
                      0.566 ms 43.85%  [unaccounted]
                  0.829 ms 34.07%  dead_store_elimination [12 x  69.082 us]
                    308.752 us 37.24%  live_variable_analysis [12 x  25.729 us]
                    520.229 us 62.76%  [unaccounted]
                  0.205 ms  8.41%  die                   [12 x  17.047 us]
                  0.109 ms  4.47%  [unaccounted]
          0.006 ms  0.00%  get_this_thread_runtime_module [12 x 536.442 ns]
          0.484 ms  0.10%  eliminate_unused_functions [12 x  40.332 us]
        279.041 ms 60.46%  optimize_module       [12 x  23.253 ms]
             15.124 ms  5.42%  llvm_function_pass    [12 x   1.260 ms]
            250.798 ms 89.88%  llvm_module_pass      [12 x  20.900 ms]
             13.118 ms  4.70%  [unaccounted]
        150.920 ms 32.70%  [unaccounted]
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
[E 01/08/25 11:20:51.768 195653] [struct_llvm.cpp:run@266] Assertion failure: (int)snodes.size() <= taichi_max_num_snodes

```

Huisouan, 2025-01-08:
In addition, when you generate more than 20 sub-terrains at once, the computational cost of terrain simplification by the open3d library will be extremely high, taking several minutes to simplify the network.

mi-robotics, 2025-01-08:
@Huisouan Do you have any ideas on how to extract the terrain in the rigid solver so we dont duplicate

mi-robotics, 2025-01-09:
@Huisouan @zhouxian I think i have a temporary solution that works. it is working with large number of envs and large tarrains.

I make some assume there is only one terrain and that terrains are build first. 
I would appreciate feedback on this and any ideas for more a elegant solution.

Before going into, i agree with @Huisouan that it seems sensible to not duplicate the terrain, or other static geoms for each env. I looked into doing this, but do not understand the system well enough to make those changes. 

However, i found that the biggest issues i encountered was due to the explosion of the verts_state in the rigid rigid body solver.

I simply isolate the terrain verts and create a seperate vert state tensor that is not duplicated across envs. 

the changes are as follows:

Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py

```
def build(self):
        self.n_envs = self.sim.n_envs
        self._B = self.sim._B
        self._para_level = self.sim._para_level

        for entity in self._entities:
            entity._build()

        self._n_qs = self.n_qs
        self._n_dofs = self.n_dofs
        self._n_links = self.n_links
        self._n_joints = self.n_joints
        self._n_geoms = self.n_geoms
        self._n_cells = self.n_cells
        self._n_verts = self.n_verts
        self._n_faces = self.n_faces
        self._n_edges = self.n_edges
        self._n_vgeoms = self.n_vgeoms
        self._n_vfaces = self.n_vfaces
        self._n_vverts = self.n_vverts
        self._n_entities = self.n_entities

        self._geoms = self.geoms
        self._vgeoms = self.vgeoms
        self._links = self.links
        self._joints = self.joints

        # used for creating dummy fields for compilation to work
        self.n_qs_ = max(1, self.n_qs)
        self.n_dofs_ = max(1, self.n_dofs)
        self.n_links_ = max(1, self.n_links)
        self.n_joints_ = max(1, self.n_joints)
        self.n_geoms_ = max(1, self.n_geoms)
        self.n_cells_ = max(1, self.n_cells)
        self.n_verts_ = max(1, self.n_verts)
        self.n_faces_ = max(1, self.n_faces)
        self.n_edges_ = max(1, self.n_edges)
        self.n_vgeoms_ = max(1, self.n_vgeoms)
        self.n_vfaces_ = max(1, self.n_vfaces)
        self.n_vverts_ = max(1, self.n_vverts)
        self.n_entities_ = max(1, self.n_entities)

        ################ CHANGES ##############
        terrains = []
        for e in self.entities:
            if isinstance(e._morph , gs.morphs.Terrain ):
                terrains.append(e)
        self.n_terrain_verts_ = terrains[0].n_verts
        ################ CHANGES ##############

        if self.is_active():
            self._init_mass_mat()
            self._init_dof_fields()
            self._init_vert_fields()
[... 124 lines cut ...]

                for i_v in range(g_info.vert_start, g_info.vert_end):
                    lower = ti.min(lower, self.verts_state[i_v-self.n_terrain_verts_, i_b].pos)
                    upper = ti.max(upper, self.verts_state[i_v-self.n_terrain_verts_, i_b].pos)

                self.geoms_state[i_g, i_b].aabb_min = lower
                self.geoms_state[i_g, i_b].aabb_max = upper
            
            else:
        
                lower = self.terrain_verts_state[g_info.vert_start, 0].pos
                upper = self.terrain_verts_state[g_info.vert_start, 0].pos

                for i_v in range(g_info.vert_start, g_info.vert_end):
                    lower = ti.min(lower, self.terrain_verts_state[i_v, 0].pos)
                    upper = ti.max(upper, self.terrain_verts_state[i_v, 0].pos)

                self.geoms_state[i_g, i_b].aabb_min = lower
                self.geoms_state[i_g, i_b].aabb_max = upper

```

you can see here my collision seem good:

[image]

Huisouan, 2025-01-10:
Thank you very much for your reply. I will immediately try to improve the generation of scenes, and perhaps we can publish a request to help the authors improve Genesis.
Then，@zhouxian .

PR #673 (2025-02-07): [FEATURE] Support fixed entity
## Description
Vertices of fixed entity (terrain for example) will not be batched to reduce memory cosumption.

## Related Issue
Resolve #498.

## Motivation and Context
See #498.

## How Has This Been Tested?
Run examples/rigid/terrain_subterrain.py
closed 2025-02-07 by ziyanx02