Genesis-Embodied-AI/genesis-world #500: Assertion failed: (p != nullptr), function bind_pipeline, file metal_device.mm, line 382.

ZeroSaturn, 2025-01-07:
https://github.com/user-attachments/assets/5705adc1-13f1-48f1-ad2e-82985266911f

MAC OS M1

I have been adjusting the code (CHATGPT) to allow for rendering some of the example files since the viewers not supported on Mac, I got elastic dragon and pbd_liquid to work, i linked my dragon and liquid render above, but the code im attatching below produces the error , its originaly from  pbd_cloth. I've seen this error posted before but I hope that I could possibly give some more insight into the cause of the problem, 

Here is the code I ran to get the error :

`import argparse
import numpy as np
import genesis as gs

def main():
    parser = argparse.ArgumentParser()
    parser.add_argument("-v", "--vis", action="store_true", default=False)
    parser.add_argument("-c", "--cpu", action="store_true", default=False)
    args = parser.parse_args()

    ########################## init ##########################
    gs.init(backend=gs.cpu if args.cpu else gs.gpu, logging_level="debug")

    ########################## create a scene ##########################

    scene = gs.Scene(
        sim_options=gs.options.SimOptions(
            substeps=10,
            gravity=(0, 0, -9.8),
        ),
        viewer_options=gs.options.ViewerOptions(
            camera_pos=(2, 2, 1.5),
            camera_lookat=(0, 0, 0.5),
            camera_up=(0, 0, 1),
        ),
        show_viewer=args.vis,
    )

    ########################## materials ##########################
    mat_elastic = gs.materials.PBD.Elastic()

    ########################## entities ##########################

    bunny = scene.add_entity(
        material=mat_elastic,
        morph=gs.morphs.Mesh(
            file="meshes/dragon/dragon.obj",
            scale=0.003,
            pos=(0, 0, 0.8),
        ),
        surface=gs.surfaces.Default(
            # vis_mode='recon',
        ),
    )

    ########################## add camera ##########################
    cam = scene.add_camera(
        res=(1280, 720),  # You can adjust the resolution as needed
        pos=(2, 2, 1.5),
        lookat=(0, 0, 0.5),
        fov=60,  # Field of view in degrees
        GUI=False,  # Disable GUI for recording
    )

    ########################## build ##########################
    scene.build()

    ########################## start recording ##########################
    cam.start_recording()

    ########################## run simulation and record ##########################
    horizon = 150
    for i in range(horizon):
        scene.step()
        # Optionally, update camera pose if you want dynamic camera movement
        # Example: Rotate the camera around the z-axis
        angle = (i / horizon) * 2 * np.pi
        cam.set_pose(
            pos=(2 * np.cos(angle), 2 * np.sin(angle), 1.5),
            lookat=(0, 0, 0.5),
        )
        cam.render()

    ########################## stop recording and save video ##########################
    cam.stop_recording(save_to_filename='output_video.mp4', fps=60)

if __name__ == "__main__":
    main()
`
produces the following error:
`RHI Error: cannot create compute pipeline state: Compiler encountered an internal error (code=3)
RHI Error: cannot create compute pipeline state: Compiler encountered an internal error (code=3)
Assertion failed: (p != nullptr), function bind_pipeline, file metal_device.mm, line 382.

Process finished with exit code 134 (interrupted by signal 6:SIGABRT)`

hckang-dev, 2025-01-15:
I commented out this line and it worked.

Genesis/genesis/engine/solvers/rigid/constraint_solver_decomp.py
line 840:  self.efc_force[i_c, i_b] = -self.efc_D[i_c, i_b] * self.Jaref[i_c, i_b] * self.active[i_c, i_b]

duburcqa, 2025-03-22:
Could you try on main branch ?

closed 2025-05-24 by duburcqa