Genesis-Embodied-AI/genesis-world #504: CMake version error

sickopickle, 2025-01-08:
Upon running
```
 cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF 
cmake --build build -j $(nproc)

```
I get the following error 
```
CMake Error at src/compute/CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.26 or higher is required.  You are running version 3.22.1
```

I followed the instructions 
```
# if your system's cmake version is under 3.18, uninstall that and reinstall via snap
sudo snap install cmake --classic
```

sickopickle, 2025-01-08:
I followed the instructions and got
```

CMake Warning at src/compute/src/backends/cuda/CMakeLists.txt:149 (message):
  NVTT not found! Please install NVTT from
  https://developer.nvidia.com/nvidia-texture-tools-exporter and set the
  CMake NVTT_DIR variable to the folder containing nvtt*.dll (e.g.
  C:\Program Files\NVIDIA Corporation\NVIDIA Texture Tools) .

CMake Warning at src/compute/src/backends/cuda/CMakeLists.txt:156 (message):
  NVTT_DIR didn't contain an NVTT shared library of the form nvtt*.dll or
  libnvtt.so.*! Is NVTT_DIR set correctly? NVTT_DIR was

CMake Warning (dev) at /home/sickopickle/.local/share/cmake/pybind11/FindPythonLibsNew.cmake:98 (find_package):
  Policy CMP0148 is not set: The FindPythonInterp and FindPythonLibs modules
  are removed.  Run "cmake --help-policy CMP0148" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

Call Stack (most recent call first):
  /home/sickopickle/.local/share/cmake/pybind11/pybind11Tools.cmake:50 (find_package)
  /home/sickopickle/.local/share/cmake/pybind11/pybind11Common.cmake:206 (include)
  /home/sickopickle/.local/share/cmake/pybind11/pybind11Config.cmake:250 (include)
  src/apps/CMakeLists.txt:12 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found pybind11: /home/sickopickle/.local/include (found version "2.9.1")
CMake Error at /usr/local/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Python (missing: Python_EXECUTABLE Python_INCLUDE_DIRS
  Python_LIBRARIES Interpreter Development Development.Module
  Development.Embed) (Required is exact version "3.9")

      Reason given by package: 
          Interpreter: Wrong version for the interpreter "/bin/python"

Call Stack (most recent call first):
  /usr/local/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.31/Modules/FindPython/Support.cmake:4002 (find_package_handle_standard_args)
  /usr/local/share/cmake-3.31/Modules/FindPython.cmake:631 (include)
  src/apps/CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
gmake: Makefile: No such file or directory
gmake: *** No rule to make target 'Makefile'.  Stop.

```

zswang666, 2025-01-08:
Could you check which cmake and python you are using and see if those are as expected?
```
which cmake
which python
```

Also are you using virtual env for python?

sickopickle, 2025-01-08:
```
which cmake
/usr/local/bin/cmake

which python
/usr/bin/python

```
I'm not using a virtual env for python.

zswang666, 2025-01-11:
It seems like you are not using the cmake that you installed with snap as 
```
sudo snap install cmake --classic
```
Could you check `/snap/bin/cmake`?

For python version, it should work with at least 3.9 and 3.10 as they both work on my end.

sickopickle, 2025-01-12:
```
/snap/cmake/1435
/snap/cmake/1437
/snap/cmake/current
```

sickopickle, 2025-01-15:
So I fixed it by putting all my faith in ChatGPT and completely removing my faulty CMake installations and reinstalling it with its instructions. I think I messed up the Kitware CMake installation originally by not following the instructions properly.

[image]

closed 2025-01-15 by sickopickle