Genesis-Embodied-AI/genesis-world #513: Cannot install genesis from Python 3.12 due to Taichi

mulliganaceous, 2025-01-08:
I created a new Python environment using conda. However, I am not able to install Genesis while with the latest version of Python (despite saying Python 3.9) because there is no matchin distribution found for taichi.

```
(genesis) [samaria@mulliganaceous Genesis]$ pip3 --version
pip 24.3.1 from /usr/lib/python3.13/site-packages/pip (python 3.13)
(genesis) [samaria@mulliganaceous Genesis]$ pip install genesis-world --break-system-packages
Defaulting to user installation because normal site-packages is not writeable
Collecting genesis-world
  Using cached genesis_world-0.2.1-py3-none-any.whl.metadata (13 kB)
Requirement already satisfied: psutil in /usr/lib/python3.13/site-packages (from genesis-world) (6.1.1)
Collecting scikit-image (from genesis-world)
  Using cached scikit_image-0.25.0-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Using cached genesis_world-0.2.0-py3-none-any.whl.metadata (11 kB)
  Using cached genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1, genesis-world==0.2.0 and genesis-world==0.2.1 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.1 depends on taichi==1.7.3
    genesis-world 0.2.0 depends on taichi==1.7.2
    genesis-world 0.1.1 depends on taichi==1.7.2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
(genesis) [samaria@mulliganaceous Genesis]$ pip install -e . --break-system-packages
Defaulting to user installation because normal site-packages is not writeable
Obtaining file:///home/samaria/S/GitHub/Genesis
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: psutil in /usr/lib/python3.13/site-packages (from genesis-world==0.2.1) (6.1.1)
Collecting scikit-image (from genesis-world==0.2.1)
  Using cached scikit_image-0.25.0-cp313-cp313-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
ERROR: Could not find a version that satisfies the requirement taichi==1.7.3 (from genesis-world) (from versions: none)
ERROR: No matching distribution found for taichi==1.7.3
(genesis) [samaria@mulliganaceous Genesis]$ pip3 install taichi --break-system-packages
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement taichi (from versions: none)
ERROR: No matching distribution found for taichi
(genesis) [samaria@mulliganaceous Genesis]$ 

```

ai-bits, 2025-01-09:
taichi is not compatible with Python 3.13, but 3.12(.8) workes for me on any platform. (down to 3.10, Mac not fully 3.9)
Be sure to install PyTorch FIRST and THEN Genesis (best from the repo to get the latest updates) to avoid numpy (0.2.0.1) problems.
See my "visualizer lottery" threads for more info.
Cheers
G.

duburcqa, 2025-03-23:
Documentation has been updated. It will be reflected at next release.

closed 2025-03-23 by duburcqa

cielavenir, 2025-07-26:
Hi there, I put `taichi-1.7.3-cp313-cp313-manylinux_2_31_x86_64.whl` https://github.com/taichi-dev/taichi/pull/8749#issuecomment-3121480456 . Try if you are interested.

cielavenir, 2025-07-30:
Rebuilt taichi using clang-11 https://github.com/taichi-dev/taichi/pull/8749#issuecomment-3134892445