Genesis-Embodied-AI/genesis-world #515: OpenGL error

ZhihongLiao, 2025-01-09:
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/root/miniconda3/envs/myconda/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/root/miniconda3/envs/myconda/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1161, in _init_and_start_app
    pyglet.clock.tick()
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 955, in _time_event
    self.on_draw()
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._render()
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1098, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/genesis/ext/pyrender/texture.py", line 219, in _add_to_context
    max_aniso = glGetFloat(texture_filter_anisotropic.GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT)
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/OpenGL/wrapper.py", line 678, in wrapperCall
    raise err
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/OpenGL/wrapper.py", line 671, in wrapperCall
    result = wrappedOperation( *cArguments )
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
  File "/root/miniconda3/envs/myconda/lib/python3.10/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glGetFloatv,
	pyArgs = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		<object object at 0x7f076d7e45e0>,
	),
	cArgs = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		array([0.], dtype=float32),
	),
	cArguments = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		array([0.], dtype=float32),
	)
)

wuzeyuuu, 2025-01-14:
same to me, I used vnc mode

lnotspotl, 2025-01-25:
This worked for me:

```
pip3 install PyOpenGL==3.1.4
```

duburcqa, 2025-02-27:
Duplicate of https://github.com/Genesis-Embodied-AI/Genesis/issues/142. Closing.

closed 2025-04-25 by duburcqa