Genesis-Embodied-AI/genesis-world #516: *** stack smashing detected ***: terminated

liuwuwuwu, 2025-01-09:
运行https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html 中 Camera & Headless Rendering 的demo时报错：

[Genesis] [23:30:37] [INFO] Building scene <9e58985>...
[W 01/08/25 23:30:37.658 51946] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/08/25 23:30:38.042 51946] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/08/25 23:30:40.854 51946] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[Genesis] [23:30:40] [INFO] Compiling simulation kernels...
[Genesis] [23:30:45] [INFO] Building visualizer...
*** stack smashing detected ***: terminated
Aborted (core dumped)

运行环境CPU
opencv版本4.10.0
想问下这是什么原因

tikongheyue, 2025-01-18:
我和你一样，蹲一下解决方法

StillRunDeep, 2025-02-11:
mark
same issue

duburcqa, 2025-03-22:
Could you try on main branch ?