Genesis-Embodied-AI/genesis-world #521: Invalid device ID (0)

Dragon2938734, 2025-01-09:
i install the genesis as instructor, and then run the hello world script, but it failed to find valid device. Is there anyone meet this case?

[image]

Dragon2938734, 2025-01-09:
I have set "gs.init(backend=gs.gpu)", and run on a server with 8 4090 gpu. (python=3.0; torch=2.0.0; cuda = 11.8)

Kashu7100, 2025-01-09:
If you are running on headless environment, run genesis with `show_viewer=False`.

Dragon2938734, 2025-01-09:
> If you are running on a headless environment, run genesis with `show_viewer=False`.

yeah, i have set "show_viewer=False", and then i meet this problem .

Kashu7100, 2025-01-09:
Check #474

HxLyn3, 2025-02-12:
I attempted to force Pyrender to use OSMesa for rendering, and it resolved the issue.
```bash
sudo apt-get install libosmesa6
export PYOPENGL_PLATFORM=osmesa
python your_script.py
```