Genesis-Embodied-AI/genesis-world #522: Slow video rendering

goncalog, 2025-01-09:
Hello, thank you for the amazing work!

Was wondering if you could suggest ways to speed-up this video rendering (the it's based on https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html)?
```
import genesis as gs
gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer = False,
    renderer=gs.renderers.Rasterizer(),
)

plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

cam = scene.add_camera(
    res    = (640, 480),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False,
)

scene.build()

# render rgb, depth, segmentation, and normal
# rgb, depth, segmentation, normal = cam.render(rgb=True, depth=True, segmentation=True, normal=True)

cam.start_recording()
import numpy as np

for i in range(5):
    scene.step()
    cam.set_pose(
        pos    = (3.0 * np.sin(i / 60), 3.0 * np.cos(i / 60), 2.5),
        lookat = (0, 0, 0.5),
    )
    cam.render()
cam.stop_recording(save_to_filename='video.mp4', fps=60)
```

and this is the output:
```
[Genesis] [17:12:55] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [17:12:55] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [17:12:55] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [17:12:55] [INFO] Running on [Tesla T4] with backend gs.cuda. Device memory: 14.74 GB.
[Genesis] [17:12:56] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [17:12:57] [INFO] Scene <58a6761> created.
[Genesis] [17:12:57] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <281daf3>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [17:12:57] [INFO] Preprocessing geom idx 0.
[Genesis] [17:12:57] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <9b48618>, morph: <gs.morphs.MJCF(file='/usr/local/lib/python3.10/dist-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [17:12:58] [INFO] Preprocessing geom idx 1.
[Genesis] [17:13:01] [INFO] Preprocessing geom idx 2.
[Genesis] [17:13:03] [INFO] Preprocessing geom idx 3.
[Genesis] [17:13:06] [INFO] Preprocessing geom idx 4.
[Genesis] [17:13:08] [INFO] Preprocessing geom idx 5.
[Genesis] [17:13:11] [INFO] Preprocessing geom idx 6.
[Genesis] [17:13:12] [INFO] Preprocessing geom idx 7.
[Genesis] [17:13:12] [INFO] Preprocessing geom idx 8.
[Genesis] [17:13:12] [INFO] Preprocessing geom idx 9.
[Genesis] [17:13:13] [INFO] Preprocessing geom idx 10.
[Genesis] [17:13:14] [INFO] Preprocessing geom idx 11.
[Genesis] [17:13:14] [INFO] Preprocessing geom idx 12.
[Genesis] [17:13:14] [INFO] Preprocessing geom idx 13.
[Genesis] [17:13:15] [INFO] Preprocessing geom idx 14.
[Genesis] [17:13:15] [INFO] Preprocessing geom idx 16.
[Genesis] [17:13:15] [INFO] Building scene <58a6761>...
[Genesis] [17:13:28] [INFO] Compiling simulation kernels...
[Genesis] [17:14:20] [INFO] Building visualizer...
[Genesis] [17:15:26] [INFO] Running at 6544.48 FPS.
[Genesis] [17:15:54] [INFO] Running at 0.70 FPS.
[Genesis] [17:17:55] [INFO] Running at 0.14 FPS.
[Genesis] [17:18:26] [INFO] Running at 0.12 FPS.
[Genesis] [17:19:05] [INFO] Running at 0.10 FPS.
[Genesis] [17:28:28] [INFO] Saving video to video.mp4.
[Genesis] [17:28:28] [INFO] Video saved.
```

thank you!

ziyanx02, 2025-01-09:
I believe this is the same issue as #138, where they encountered the same problem on a Tesla L40. It is likely caused by an incompatible graphics device or driver.

goncalog, 2025-01-09:
thank you! i can try again tomorrow on a different machine and install new drivers - which one should i use?

and just fwiw, the sim runs much faster without rendering/viewer

ziyanx02, 2025-01-09:
Maybe it's better to try docker first. @Kashu7100 do you have any idea?

goncalog, 2025-01-09:
thank you, will do! thought you were referring to the nvidia drivers but i don't see any drivers referred to in the Dockerfile - did i miss them?

wangyian-me, 2025-01-09:
I think you might want to refer to issue #230 #203 .

Kashu7100, 2025-01-10:
Can you check 12.1 with 535 or 12.4 with 550?

goncalog, 2025-01-10:
awesome, thank you! will try that

goncalog, 2025-01-10:
> I think you might want to refer to issue #230 #203 .

this worked - thank you!

closed 2025-01-10 by goncalog