Genesis-Embodied-AI/genesis-world #524: CPU bottleneck

DTSwink, 2025-01-09:
after profiling i found out 1 CPU is used at 99% while the others arent working, and the gpu is barely working too
profiling go2_env for RL training shows that the 2 slowest function are `torch.as_tensor` and  `_validate_2D_io_variables` which seem to be called several time at every step
is it intended ? thanks

```

3440    0.008    0.000    3.147    0.001 /home/singerie/Documents/rsl_rl/rsl_rl/modules/actor_critic.py:134(act)
     3200    5.781    0.002    5.781    0.002 {method 'run_backward' of 'torch._C._EngineBase' objects}
     3200    0.015    0.000    5.800    0.002 /home/singerie/Documents/myenv2/lib/python3.12/site-packages/torch/autograd/graph.py:816(_engine_run_backward)
     3200    0.019    0.000    5.943    0.002 /home/singerie/Documents/myenv2/lib/python3.12/site-packages/torch/autograd/__init__.py:242(backward)
     3200    0.019    0.000    5.963    0.002 /home/singerie/Documents/myenv2/lib/python3.12/site-packages/torch/_tensor.py:525(backward)
        1    8.908    8.908    8.908    8.908 {built-in method builtins.input}
    42760   18.538    0.000   18.538    0.000 {built-in method torch.as_tensor}
    35721    0.268    0.000   19.898    0.001 /home/singerie/Documents/Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py:3205(_validate_2D_io_variables)
```

bolkhair, 2025-01-11:
3200    5.781    0.002    5.781    0.002 {method 'run_backward' of 'torch._C._EngineBase' objects}
     3200    0.015    0.000    5.800    0.002 /home/me/Documents/myenv2/lib/python3.12/site-packages/torch/autograd/graph.py:816(_engine_run_backward)
     3200    0.019    0.000    5.943    0.002 /home/me/Documents/myenv2/lib/python3.12/site-packages/torch/autograd/__init__.py:242(backward)
     3200    0.019    0.000    5.963    0.002 /home/me/Documents/myenv2/lib/python3.12/site-packages/torch/_tensor.py:525(backward)
        1    8.908    8.908    8.908    8.908 {built-in method builtins.input}
    42760   18.538    0.000   18.538    0.000 {built-in method torch.as_tensor}
    35721    0.268    0.000   19.898    0.001 /home/me/Documents/Genesis/genesis/engine/solvers/rigid/rigid_solver_decomp.py:3205(_validate_2D_io_variables)

duburcqa, 2025-03-22:
Do you have any reproduction script ?

duburcqa, 2025-05-24:
Profiling is not faithful because of GPU synchronisation going into the way. This does not represent the reality.