Genesis-Embodied-AI/genesis-world #528: can't unbox array from PyObject into native value.  The object maybe of a different type

leileicheng854, 2025-01-10:
raceback (most recent call last):
  File "/home/qiu/miniconda3/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/qiu/miniconda3/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/media/qiu/D/chengleijun/code/Genesis/genesis/ext/pyrender/viewer.py", line 1185, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/qiu/miniconda3/lib/python3.10/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
  File "/home/qiu/miniconda3/lib/python3.10/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/qiu/miniconda3/lib/python3.10/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/media/qiu/D/chengleijun/code/Genesis/genesis/ext/pyrender/viewer.py", line 979, in _time_event
    self.on_draw()
  File "/media/qiu/D/chengleijun/code/Genesis/genesis/ext/pyrender/viewer.py", line 674, in on_draw
    self.update_buffers()
  File "/media/qiu/D/chengleijun/code/Genesis/genesis/ext/pyrender/viewer.py", line 634, in update_buffers
    self._renderer.jit.update_buffer(self.pending_buffer_updates)
  File "/media/qiu/D/chengleijun/code/Genesis/genesis/ext/pyrender/jit_render.py", line 852, in update_buffer
    self._update_buffer(updates, self.gl.wrapper_instance)
TypeError: can't unbox array from PyObject into native value.  The object maybe of a different type

Riften, 2025-01-22:
Same error when running in docker environment. Did you solve it?

ChenyuZhangX, 2025-03-20:
Same error, I installed it following the official documentation and using a new conda environment.

duburcqa, 2025-03-20:
Could you try installing the main branch of Genesis ?

closed 2025-05-24 by duburcqa