Genesis-Embodied-AI/genesis-world #529: Unable to load glb glb format sence

qZhang88, 2025-01-10:
I have down load scene from habitat-sim test examples and load with following code, but nothing show up except the empty ground plane.

``` python
import genesis as gs
gs.init(backend=gs.cpu)
 
scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
apartment = scene.add_entity(gs.morphs.Mesh(file="./habitat_test_scenes/apartment_1.glb"))

scene.build()

for i in range(1000):
    scene.step()
```

There are some warning btw
[Genesis] [14:50:54] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <89f1544>, morph: <gs.morphs.Mesh(file='./habitat_test_scenes/apartment_1.glb')>, material: <gs.materials.Rigid>.
/home/x/miniconda3/envs/genesis/lib/python3.11/site-packages/pygltflib/__init__.py:830: UserWarning: pygltflib currently does not remove image data from the buffer when converting to data uri.Please open an issue at https://gitlab.com/dodgyville/pygltflib/issues
  warnings.warn("pygltflib currently does not remove image data "
[Genesis] [14:50:56] [WARNING] Texture given but asset missing uv info (or failed to load).
[Genesis] [14:50:59] [INFO] Preprocessing geom idx 1.
[Genesis] [14:58:21] [INFO] Building scene <9bceb42>...
[W 01/10/25 14:58:21.698 329694] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/10/25 14:58:22.588 329694] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.
[W 01/10/25 14:58:31.421 329694] [frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed.

qZhang88, 2025-01-11:
glb could download following instruction in https://github.com/facebookresearch/habitat-sim

```
python -m habitat_sim.utils.datasets_download --uids habitat_test_scenes --data-path /path/to/data/
```

spomichter, 2025-02-11:
@qZhang88 If this is still an open issue would try with these parameters

```python
office = scene.add_entity(
    gs.morphs.Mesh(
        file="../assets/office.glb",
        fixed=True,  # Make it static
        euler=(-90, 180, 0),  # only if needed
        pos=(0, 0, 2.0)  # only if needed
    )
)
```

qZhang88, 2025-02-11:
> [@qZhang88](https://github.com/qZhang88) If this is still an open issue would try with these parameters
> 
> office = scene.add_entity(
>     gs.morphs.Mesh(
>         file="../assets/office.glb",
>         fixed=True,  # Make it static
>         euler=(-90, 180, 0),  # only if needed
>         pos=(0, 0, 2.0)  # only if needed
>     )
> )

tried again (with and without euler and pos), still not loaded, and nothing show up.

debOliveira, 2025-03-16:
@qZhang88, do you have an update on this issue? I am also interested in importing glb files.

duburcqa, 2025-05-24:
Probably related to https://github.com/Genesis-Embodied-AI/Genesis/issues/955

Riften, 2025-06-25:
For anybody interested, `gltf-transform` now supports decompress KTX2 textures in glb file.

See https://github.com/3dlg-hcvc/hssd/issues/25#issuecomment-3004660306