Genesis-Embodied-AI/genesis-world #530: In the drone project，the hover_eval.py is extremely slow

zhrli, 2025-01-10:
[image]

KafuuChikai, 2025-01-10:
你可以尝试修改一下`hover_eval.py`?

- 把`env_cfg["visualize_target"] = True`这一行注释
- 删去参数`--record`

以上两点是与locomotion不同的feature，其他应该都是一致的。然后运行

```bash
python hover_eval.py -e drone-hovering --ckpt 500
```

如果还是不行，可以试试不使用docker？

KafuuChikai, 2025-01-10:
可以提供一下你在traing时候的数据吗？

我看看在training的时候计算效率是否与我的一致，我的测试平台：

- ubuntu 20.04
- AMD Ryzen 7950x
- RTX 4090 D
- DDR5 64G

1分30秒左右完成训练

zhrli, 2025-01-10:
> * env_cfg["visualize_target"]

[image]
正常速度了

KafuuChikai, 2025-01-10:
> > * env_cfg["visualize_target"]
> 
> ![image](https://private-user-images.githubusercontent.com/11074703/401877104-bc2b96f0-27e6-4d31-a93c-1697f0f8816f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY0OTYwMzksIm5iZiI6MTczNjQ5NTczOSwicGF0aCI6Ii8xMTA3NDcwMy80MDE4NzcxMDQtYmMyYjk2ZjAtMjdlNi00ZDMxLWE5M2MtMTY5N2YwZjg4MTZmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEwVDA3NTUzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVjYTgxMDFiMzc4ZTk2ZmQwNmQ2NzQ5OWZhZWZhNTkyZGJiNjhkNWY3YjIwYjRjN2NhMjI5MDVmMDk3OTc4MjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jvT72dbfTmgOt7MKv2rGeK3JblstC41yKK8BtN3625o) 正常速度了
> 
> 加 --record 就慢了

所以是`--record`导致的吗？

**我提供以下两个方向排查具体问题：**

1. 我写的视频录制功能是参考`examples/tutorials/visualization.py`，你看看这个是否能正常运行？

2. 在`hover_eval.py`里，你可以查看一下代码块
  
    ```python
        with torch.no_grad():
            if args.record:
                env.cam.start_recording()
                for _ in range(max_sim_step):
                    actions = policy(obs)
                    obs, _, rews, dones, infos = env.step(actions)
                    env.cam.render()
                env.cam.stop_recording(save_to_filename="video.mp4", fps=env_cfg["max_visualize_FPS"])
            else:
                for _ in range(max_sim_step):
                    actions = policy(obs)
                    obs, _, rews, dones, infos = env.step(actions)
    ```
    
    record的flag在这里触发，你可以看一下具体是哪一行导致效率变低？

zhrli, 2025-01-10:
> 可以提供一下你在traing时候的数据吗？
> 
> 我看看在training的时候计算效率是否与我的一致，我的测试平台：
> 
> * ubuntu 20.04
> * AMD Ryzen 7950x
> * RTX 4090 D
> * DDR5 64G
> 
> 仿真速度1,200,000 step/s, 1分30秒左右完成训练

base) lizhaorui@ubuntu:~/DL/Genesis/examples$ nvidia-smi
Fri Jan 10 16:02:14 2025       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.113.01             Driver Version: 535.113.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 4090        Off | 00000000:18:00.0 Off |                  Off |
| 30%   30C    P2              57W / 450W |    652MiB / 24564MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 4090        Off | 00000000:C3:00.0  On |                  Off |
| 58%   31C    P8              31W / 450W |    221MiB / 24564MiB |     17%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A   1749295      G   /usr/lib/xorg/Xorg                            4MiB |
|    0   N/A  N/A   2623347      C   python                                      632MiB |
|    1   N/A  N/A   1749295      G   /usr/lib/xorg/Xorg                          209MiB |
+---------------------------------------------------------------------------------------+
(base) lizhaorui@ubuntu:~/DL/Genesis/examples$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Jun_13_19:16:58_PDT_2023
Cuda compilation tools, release 12.2, V12.2.91
Build cuda_12.2.r12.2/compiler.32965470_0

好像遇到我这个问题的人应该很多，有更新CUDA的 ，有加代码的 。  这个例子我在MACpro M3 上没跑起来，段错误。

KafuuChikai, 2025-01-10:
> base) lizhaorui@ubuntu:~/DL/Genesis/examples$ nvidia-smi Fri Jan 10 16:02:14 2025 +---------------------------------------------------------------------------------------+ | NVIDIA-SMI 535.113.01 [...]

我的CUDA版本比你还新，12.4

这个就和引擎有关了，还是等后续更新吧，兼容性还有待提升

formoree, 2025-01-10:
> Contributor

CUDA版本对genesis的运行效果是不是有大的影响？
我之前租用GPU，CUDA 11.8无法调用GPU进行运行，12.2以上就可以

formoree, 2025-01-10:
https://github.com/Genesis-Embodied-AI/Genesis/issues/203  具体问题是上面这个issue

zhrli, 2025-01-10:
> > Contributor
> 
> CUDA版本对genesis的运行效果是不是有大的影响？ 我之前租用GPU，CUDA 11.8无法调用GPU进行运行，12.2以上就可以

所以拉了docker    但是docker好像有的运行也不行

zhrli, 2025-01-10:
> 所以是`--record`导致的吗？
>
> **我提供以下两个方向排查具体问题：**
>
> 1. 我写的视频录制功能是参考`examples/tutorials/visualization.py`，你看看这个是否能正常运行？
> 2. 在`hover_eval.py`里，你可以查看一下代码块
>    ```python
>        with torch.no_grad():
>            if [...]

visualization不能正常运行，在linux和macos上都跑了，linux是慢，mac上报错 opencv的错  ， 估计就是cuda版本太低了

KafuuChikai, 2025-01-10:
那么可能是引擎或者docker的问题，你可以提出新的issue问问？关于camera.render() @zhouxian

closed 2025-05-24 by duburcqa